abperiasamy/rtl8812AU_8821AU_linux #29: Error in compiling

akhepcat, 2015-01-20:
Just tried to build this for 3.18.2, from a fresh git-pull, and am receiving this error:

```
In file included from /usr/src/rtl8812AU_8821AU_linux/include/drv_types.h:141:0,
                 from /usr/src/rtl8812AU_8821AU_linux/os_dep/linux/ioctl_cfg80211.c:22:
/usr/src/rtl8812AU_8821AU_linux/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_rx_p2p_action_public’:
/usr/src/rtl8812AU_8821AU_linux/include/ioctl_cfg80211.h:114:71: error: too many arguments to function ‘cfg80211_rx_mgmt’
   #define rtw_cfg80211_rx_mgmt(adapter, freq, sig_dbm, buf, len, gfp) cfg80211_rx_mgmt((adapter)->rtw_wdev, freq, sig_dbm, buf, len, 0, gfp)
                                                                       ^
```

dmix, 2015-01-22:
Same here, failed to build on 3.18.1 for me with the same error as OP.

eselkin, 2015-01-26:
Same error on 3.18.3... not that you needed any more reinforcement :-)
Though use of declared function seems correct except the struct maybe... 
bool cfg80211_rx_mgmt(struct wireless_dev *wdev, int freq, int sig_dbm, const u8 *buf, size_t len, u32 flags);

mmaps, 2015-02-09:
In 3.18 they dropped the gfp arg (the last arg to cfg80211_rx_mgmt), which results in the error. I was able to compile by making these changes suggested by Milhouse on [Kodi Tv Forums](http://forum.kodi.tv/printthread.php?tid=211501&page=2): 
```
diff -uNr '--exclude=.git' rtl8812AU_8821AU_linux_orig/include/ioctl_cfg80211.h rtl8812AU_8821AU_linux-patched/include/ioctl_cfg80211.h
--- rtl8812AU_8821AU_linux_orig/include/ioctl_cfg80211.h	2015-02-09 18:38:45.492998484 -0500
+++ rtl8812AU_8821AU_linux-patched/include/ioctl_cfg80211.h	2015-02-09 18:42:33.029120193 -0500
@@ -109,9 +109,12 @@
   #define rtw_cfg80211_rx_mgmt(adapter, freq, sig_dbm, buf, len, gfp) cfg80211_rx_mgmt((adapter)->pnetdev, freq, sig_dbm, buf, len, gfp)
 #elif (LINUX_VERSION_CODE < KERNEL_VERSION(3,12,0))
   #define rtw_cfg80211_rx_mgmt(adapter, freq, sig_dbm, buf, len, gfp) cfg80211_rx_mgmt((adapter)->rtw_wdev, freq, sig_dbm, buf, len, gfp)
-#else
+#elif (LINUX_VERSION_CODE < KERNEL_VERSION(3,18,0))
   // 3.12 added a flags argument which is just set to zero
   #define rtw_cfg80211_rx_mgmt(adapter, freq, sig_dbm, buf, len, gfp) cfg80211_rx_mgmt((adapter)->rtw_wdev, freq, sig_dbm, buf, len, 0, gfp)
+#else
+  // 3.18 dropped the gfp argument
+  #define rtw_cfg80211_rx_mgmt(adapter, freq, sig_dbm, buf, len, gfp) cfg80211_rx_mgmt((adapter)->rtw_wdev, freq, sig_dbm, buf, len, 0)
 #endif
 
 #if (LINUX_VERSION_CODE < KERNEL_VERSION(3,4,0))  && !defined(COMPAT_KERNEL_RELEASE)
```

akhepcat, 2015-02-12:
+1 to mmap's patch.

closed 2015-02-12 by akhepcat