abperiasamy/rtl8812AU_8821AU_linux #32: Compilation fails in 3.18.5 and .6

mikeholden99, 2015-02-15:
Looks like the parameters to cfg80211_rx_mgmt have changed again again :-)

The flags parameter added in 3.12 is still there, but the gfp parameter has been removed:

[root@hagrid kernels]# grep -A 1 "bool cfg80211_rx_mgmt" */include/net/cfg80211.h
3.17.8-200.fc20.x86_64/include/net/cfg80211.h:bool cfg80211_rx_mgmt(struct wireless_dev *wdev, int freq, int sig_dbm,
3.17.8-200.fc20.x86_64/include/net/cfg80211.h-		      const u8 *buf, size_t len, u32 flags, gfp_t gfp);
--
3.18.5-101.fc20.x86_64/include/net/cfg80211.h:bool cfg80211_rx_mgmt(struct wireless_dev *wdev, int freq, int sig_dbm,
3.18.5-101.fc20.x86_64/include/net/cfg80211.h-		      const u8 *buf, size_t len, u32 flags);
--
3.18.6-100.fc20.x86_64/include/net/cfg80211.h:bool cfg80211_rx_mgmt(struct wireless_dev *wdev, int freq, int sig_dbm,
3.18.6-100.fc20.x86_64/include/net/cfg80211.h-		      const u8 *buf, size_t len, u32 flags);

This will thus need a fix for include/ioctl_cfg80211.h for the #define

Razican, 2015-02-21:
I have those kind of issues also in Raspbian kernel 3.18.7:

```
/root/wifi_driver/os_dep/linux/os_intfs.c:1695:2: warning: initialization from incompatible pointer type [enabled by default]
/root/wifi_driver/os_dep/linux/os_intfs.c:1695:2: warning: (near initialization for ‘rtw_netdev_ops.ndo_select_queue’) [enabled by default]

And then:
```
/root/wifi_driver/os_dep/linux/ioctl_cfg80211.c: In function ‘cfg80211_rtw_add_key’:
/root/wifi_driver/os_dep/linux/ioctl_cfg80211.c:1250:3: warning: passing argument 2 of ‘_rtw_memcpy’ discards ‘const’ qualifier from pointer target type [enabled by default]
In file included from /root/wifi_driver/include/drv_types.h:32:0,
                 from /root/wifi_driver/os_dep/linux/ioctl_cfg80211.c:22:
/root/wifi_driver/include/osdep_service.h:167:13: note: expected ‘void *’ but argument is of type ‘const u8 *’
/root/wifi_driver/os_dep/linux/ioctl_cfg80211.c:1256:3: warning: passing argument 2 of ‘_rtw_memcpy’ discards ‘const’ qualifier from pointer target type [enabled by default]
In file included from /root/wifi_driver/include/drv_types.h:32:0,
                 from /root/wifi_driver/os_dep/linux/ioctl_cfg80211.c:22:
/root/wifi_driver/include/osdep_service.h:167:13: note: expected ‘void *’ but argument is of type ‘const u8 *’
/root/wifi_driver/os_dep/linux/ioctl_cfg80211.c: In function ‘cfg80211_rtw_connect’:
/root/wifi_driver/os_dep/linux/ioctl_cfg80211.c:2550:2: warning: passing argument 2 of ‘_rtw_memcpy’ discards ‘const’ qualifier from pointer target type [enabled by default]
In file included from /root/wifi_driver/include/drv_types.h:32:0,
                 from /root/wifi_driver/os_dep/linux/ioctl_cfg80211.c:22:
/root/wifi_driver/include/osdep_service.h:167:13: note: expected ‘void *’ but argument is of type ‘const u8 *’
/root/wifi_driver/os_dep/linux/ioctl_cfg80211.c:2587:4: warning: passing argument 2 of ‘_rtw_memcmp’ discards ‘const’ qualifier from pointer target type [enabled by default]
In file included from /root/wifi_driver/include/drv_types.h:32:0,
                 from /root/wifi_driver/os_dep/linux/ioctl_cfg80211.c:22:
/root/wifi_driver/include/osdep_service.h:168:12: note: expected ‘void *’ but argument is of type ‘const u8 *’
/root/wifi_driver/os_dep/linux/ioctl_cfg80211.c:2593:5: warning: passing argument 2 of ‘_rtw_memcmp’ discards ‘const’ qualifier from pointer target type [enabled by default]
In file included from /root/wifi_driver/include/drv_types.h:32:0,
                 from /root/wifi_driver/os_dep/linux/ioctl_cfg80211.c:22:
/root/wifi_driver/include/osdep_service.h:168:12: note: expected ‘void *’ but argument is of type ‘const u8 *’
/root/wifi_driver/os_dep/linux/ioctl_cfg80211.c:2601:14: warning: assignment discards ‘const’ qualifier from pointer target type [enabled by default]
/root/wifi_driver/os_dep/linux/ioctl_cfg80211.c:2682:2: warning: passing argument 2 of ‘rtw_cfg80211_set_wpa_ie’ discards ‘const’ qualifier from pointer target type [enabled by default]
/root/wifi_driver/os_dep/linux/ioctl_cfg80211.c:2243:12: note: expected ‘u8 *’ but argument is of type ‘const u8 *’
/root/wifi_driver/os_dep/linux/ioctl_cfg80211.c: In function ‘cfg80211_rtw_set_pmksa’:
/root/wifi_driver/os_dep/linux/ioctl_cfg80211.c:2913:2: warning: passing argument 1 of ‘_rtw_memcmp’ discards ‘const’ qualifier from pointer target type [enabled by default]
In file included from /root/wifi_driver/include/drv_types.h:32:0,
                 from /root/wifi_driver/os_dep/linux/ioctl_cfg80211.c:22:
/root/wifi_driver/include/osdep_service.h:168:12: note: expected ‘void *’ but argument is of type ‘const u8 *’
/root/wifi_driver/os_dep/linux/ioctl_cfg80211.c:2923:3: warning: passing argument 2 of ‘_rtw_memcmp’ discards ‘const’ qualifier from pointer target type [enabled by default]
In file included from /root/wifi_driver/include/drv_types.h:32:0,
                 from /root/wifi_driver/os_dep/linux/ioctl_cfg80211.c:22:
/root/wifi_driver/include/osdep_service.h:168:12: note: expected ‘void *’ but argument is of type ‘const u8 *’
/root/wifi_driver/os_dep/linux/ioctl_cfg80211.c:2927:4: warning: passing argument 2 of ‘_rtw_memcpy’ discards ‘const’ qualifier from pointer target type [enabled by default]
In file included from /root/wifi_driver/include/drv_types.h:32:0,
                 from /root/wifi_driver/os_dep/linux/ioctl_cfg80211.c:22:
/root/wifi_driver/include/osdep_service.h:167:13: note: expected ‘void *’ but argument is of type ‘const u8 *’
/root/wifi_driver/os_dep/linux/ioctl_cfg80211.c:2941:3: warning: passing argument 2 of ‘_rtw_memcpy’ discards ‘const’ qualifier from pointer target type [enabled by default]
In file included from /root/wifi_driver/include/drv_types.h:32:0,
                 from /root/wifi_driver/os_dep/linux/ioctl_cfg80211.c:22:
/root/wifi_driver/include/osdep_service.h:167:13: note: expected ‘void *’ but argument is of type ‘const u8 *’
/root/wifi_driver/os_dep/linux/ioctl_cfg80211.c:2942:3: warning: passing argument 2 of ‘_rtw_memcpy’ discards ‘const’ qualifier from pointer target type [enabled by default]
In file included from /root/wifi_driver/include/drv_types.h:32:0,
                 from /root/wifi_driver/os_dep/linux/ioctl_cfg80211.c:22:
/root/wifi_driver/include/osdep_service.h:167:13: note: expected ‘void *’ but argument is of type ‘const u8 *’
/root/wifi_driver/os_dep/linux/ioctl_cfg80211.c: In function ‘cfg80211_rtw_del_pmksa’:
[... cut ...]
make[4]: *** [/root/wifi_driver/os_dep/linux/ioctl_cfg80211.o] Error 1
/usr/src/linux-headers-3.18.0-trunk-common/Makefile:1398: recipe for target '_module_/root/wifi_driver' failed
make[3]: *** [_module_/root/wifi_driver] Error 2
Makefile:145: recipe for target 'sub-make' failed
make[2]: *** [sub-make] Error 2
Makefile:8: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-3.18.0-trunk-rpi'
Makefile:1040: recipe for target 'modules' failed
make: *** [modules] Error 2

auscompgeek, 2015-02-21:
See #18.

harshavardhana, 2015-05-04:
Should be fixed with latest merge thanks to @bits3rpent

closed 2015-05-04 by harshavardhana