abperiasamy/rtl8812AU_8821AU_linux #38: 5Ghz Issues

bits3rpent, 2015-04-21:
Hey guys, it seems that 5Ghz does not work. I am using an Alfa's AWUS036AC adapter and an ASUS AC56 to test this driver. In clarification, I can see my 5Ghz network, but cannot seem to connect to it about 80% of the time. When I do, the connection is not that stable. 

For example, my ASUS RT-AC66U Router has two networks
NetworkTheta - 2.4Ghz (this works all the time)
NetworkTheta5Ghz- 5Ghz (can always view as stated above, rarely connect)

I am actually using gnab's repo right now, and the issue I posted is right here
https://github.com/gnab/rtl8812au/issues/2

I am using Fedora 21 Workstation x86_64 with the latest kernel (at the time of this post 3.19.3-200.fc21.x86_64)

the dmesg of a successful 5Ghz connection: https://mega.co.nz/#!rVUiAYiJ!6vpJra9s8LnexQ92i5NF0YsgCKPM4C4-Z195f46btMk

will upload the dmesg of a failed connection in an edit.

Thanks in advance!

simonsmiley, 2015-04-22:
since it's a 5GHz issue I might even discuss it here, so other people can participate.
I'm using an archer T4U with pull request #40. It compiles fine, and I can connect to the 5Ghz network but only using 802.11a which makes it quite useless. I bought the adapter to use 802.11ac. I'm running archlinux with a 3.19.3 kernel

bits3rpent, 2015-04-22:
How did you find that it supports 802.11a, does your router support only using a?

simonsmiley, 2015-04-22:
I have a Netgear R7500 which supports ac. I'm connected to the 5GHz network but with a maximum bandwidth of 54 Mbits/s which makes it 802.11a It should be at at least 10 times that value with 802.11ac

bits3rpent, 2015-04-22:
I seem to be having trouble connecting to my 5Ghz network as mentioned above. /var/log/wpa_supplicant.log shows that I somehow do connect to it. What do your logs show?
wlp0s20u1: Trying to associate with 54:a0:50:58:7f:bc (SSID='NetworkTheta5Ghz' freq=5180 MHz)
wlp0s20u1: Associated with 54:a0:50:58:7f:bc
wlp0s20u1: WPA: Key negotiation completed with 54:a0:50:58:7f:bc [PTK=CCMP GTK=CCMP]
wlp0s20u1: CTRL-EVENT-CONNECTED - Connection to 54:a0:50:58:7f:bc completed [id=0 id_str=]
wlp0s20u1: CTRL-EVENT-DISCONNECTED bssid=54:a0:50:58:7f:bc reason=3 locally_generated=1

bits3rpent, 2015-04-22:
My Alfa card still cannot connect to my 5Ghz network, but my latest pull request seems to have gotten 5Ghz to work on AC56 with no issues.

simonsmiley, 2015-04-23:
I do get these errors sometimes with another card. didn't get them with this one though. it's not showing anything interesting. Just that the connection is made and what frequency...

bits3rpent, 2015-04-24:
Have you connected to your router on other devices?

Umookaep, 2015-04-27:
Same problem here, i can create 5GHz access point with hostapd. I can see the SSID but cannot connect to it. Creating a 2,4GHz access point is working fine. Maybe its a Problem of my Nexus4... i will check it with some other devices.
Used adaptor: Netis WF2190 AC 1200 / Debian Wheezy 3.16 kernel

harshavardhana, 2015-05-07:
I have been able to use this module with a 5ghz expander network without issues on Ubuntu 15.04 .

simonsmiley, 2015-05-09:
@harshavardhana and is it ac? or just a?

joseph-jja, 2015-06-19:
I have ubuntu 14.04 and kernel 3.13.0-55 and it seems to be working fine now.  I did split my 2.4 and 5 into different SSID's happ_2.4 and happy_5 ( not the real names before the _ but after it is ).  I used to have them both with the same name, but split them yesterday.  I have not had a disconnect today, nor an oops.  I also changed my 2.4 to only use the 20Mhz channel for less interference. The driver seems to work fine with my edimax 7822UAC.  Have you tried the latest?

crholm, 2015-09-29:
I did some testing of different settings in regards to this since i was experiencing some of the problems above.

I am running kernel 3.13.0-57, with a ASUS USB-AC56 and a ASUS Rt-AC66u router running 5Ghz at 802.11AC. The symptoms are problem getting it to work with dhcp, once connected (with static config) http traffic works sporadically. Some pages works fine and others did not, both using domain names or straight up ip addresses. Switching from AC to N made all problems go away so to me it seems more an issue of AC rather than 5Ghz.

There were no problems with 2.4Ghz regardless of setting

harshavardhana, 2015-09-29:
> @harshavardhana and is it ac? or just a?

I created a split network as suggested by @joseph-jja where both networks take different SSID's - it just seems cleaner and works without issues.

Polioman, 2016-01-06:
Hello,

@harshavardhana 
I use an Edimax 7811UAC dongle on a Raspberry Pi and I've got 2 questions :

1) How did you split you networks? I was not able to do it.
2) Did you manage to make it work in full AC mode? I can only make it work in A mode. Even if it is only in AC mode, I have to make it work like that.

Any help would be really appreciated.

Polioman, 2016-01-22:
Hi,

I'm having an issue with 5Ghz using this driver in AP mode. It says ''Could not set channel for kernel driver''.

I've attached usefull files : hostapd.conf, iw list and error displayed.

I don't really know if it is a driver issue or hostapd issue.
[hostapd_conf.txt](https://github.com/abperiasamy/rtl8812AU_8821AU_linux/files/100518/hostapd_conf.txt)
[hostapd_error.txt](https://github.com/abperiasamy/rtl8812AU_8821AU_linux/files/100517/hostapd_error.txt)
[iw_list.txt](https://github.com/abperiasamy/rtl8812AU_8821AU_linux/files/100519/iw_list.txt)

francoisbeaulieu, 2016-02-16:
I can't even get manage to change the channel to the 5GHz range using iwconfig. I'm on Bananian Linux. Anyone managed to get it to use a 5GHz range channel on any flavour? iwlist scan also only lists 2GHz SSIDs. The driver does report that it supports 5GHz channels.

harshavardhana, 2016-02-16:
> @harshavardhana 
> I use an Edimax 7811UAC dongle on a Raspberry Pi, It uses same driver, I've got 2 questions :
> 
> 1) How did you split you networks? I was not able to do it.
> 2) Did you manage to make it work in full AC mode? I can only make it work in A mode. Even if it is only in AC mode, I have to make it work like that.

We have never really tried these configurations - full AC mode works fine with EDIMAX-AC1200. Not sure about Raspberry Pi.

joseph-jja, 2016-02-16:
Ok, so with hostapd, the hostapd program and this driver do not work for 11ac. That's an issue with the hostapd drivers.

Polioman, 2016-02-17:
@joseph-jja 
I use this file with hostapd to have an access point in 5GHz band.
[hostapd_edimax-5GHz.txt](https://github.com/abperiasamy/rtl8812AU_8821AU_linux/files/133925/hostapd_edimax-5GHz.txt)

@francoisbeaulieu :
You can display what are capabilities of your dongle with iw list command. My dongle can work in 5GHz band but only with 20MHz or 40MHz channels (not 80 or 160).

@harshavardhana :
How did you split your network?

joseph-jja, 2016-02-17:
@Polioman 
So you don't specify a driver?  And that works?

So the splitting of the network is when you have a router like the asus ac 68u and you can setup an B/G/N and AC access points, you setup one SSID for the B/G/N and another for the AC.  I don't know if hostapd allows you to setup a single NIC as 2 different SSIDs on different frequencies.  All depends on if the wifi dongle is being used as a client connecting to an access point or a server (hostapd).

Polioman, 2016-02-18:
You can specify nl80211 for driver, or not. I removed this line since I've got issues with it.

I was not able to broadcast 2 SSID with my dongles. It works on 2,4GHz or 5GHz not both, I read [here](http://wiki.stocksy.co.uk/wiki/Multiple_SSIDs_with_hostapd) that you have to change mac address to do that and I can't with this driver (or I don't know how).

joseph-jja, 2016-02-23:
@Polioman using that config I get 11a speeds of 54Mb/s not 11ac. This is on raspian jessie

Polioman, 2016-02-23:
I've got quite the same problem but I could go up to 72Mb/s if I remind correctly. It is due to a limitation from my hardware. Mine is limited to HT20/HT40. Perhaps it is a driver limitation, can't tell.

joseph-jja, 2016-02-23:
Well you have ieee80211n commented out, probably because on the 5g channel it was not starting up.  So that means you were running 11a not 11ac or 11n on the 5g band.

Polioman, 2016-02-24:
Maybe, I wasn't able to have better bandwidth :(

tszymczak, 2017-06-19:
I am having similar issues with a Linksys WUSB6300 dongle. When I connect to the 2.4 GHz network on my router, it works fine. When I to connect to the 5 GHz network it's very unstable. The dongle connects to the network and then works normally for 10 seconds to a minute and then stops passing any network traffic. In case it matters, I have two different SSIDs for the different bands.

inconceeeivable, 2019-03-04:
Did you ever figure this out? I have the same Linksys card and am getting the same behavior on 5Ghz.