abperiasamy/rtl8812AU_8821AU_linux #39: Build failed on 3.19.3-200 kernel

bits3rpent, 2015-04-21:
/home/jiv/Downloads/rtl8812AU_8821AU_linux/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_rx_action’:
/home/jiv/Downloads/rtl8812AU_8821AU_linux/include/ioctl_cfg80211.h:114:71: error: too many arguments to function ‘cfg80211_rx_mgmt’
   #define rtw_cfg80211_rx_mgmt(adapter, freq, sig_dbm, buf, len, gfp) cfg80211_rx_mgmt((adapter)->rtw_wdev, freq, sig_dbm, buf, len, 0, gfp)
                                                                       ^
seems to be another issue in cfg80211_rx_mgmt

bits3rpent, 2015-04-21:
Fixed in pull request.

closed 2015-04-21 by bits3rpent

simonsmiley, 2015-04-22:
I know this isn't the right place, but i can't think of another way of contacting you right now.I I managed to get it to compile (running archlinux, 3.19.3-3), but the only 5GHz protocol supported seems to be 802.11a not ac... can you confirm this?

bits3rpent, 2015-04-22:
Of course, don't worry about it. You can contact me at
bits3rpent@plasmachameleon.com. As you may see in the issues section, I 
also
reported issues with 5Ghz. Haven't set my router to use 802.11a yet, but I 
will
be sure to try it!

Sent using CloudMagic 
[https://cloudmagic.com/k/d/mailapp?ct=pa&cv=6.3.14&pv=5.0]

bits3rpent, 2015-04-22:
Also, please mention the model number of your adapter :)

bits3rpent, 2015-04-22:
Another note, did you use my fork of this repo/ my pull request? If not then I guess -300 changed the parameters to cfg80211_rx_mgmt again.