abperiasamy/rtl8812AU_8821AU_linux #41: Cannot compile using Ubuntu 15.04

nzmm, 2015-04-26:
I am not terribly technical when it comes to this sort of thing but here is the last bit of output before `make` fails:

    /home/matthew/Downloads/rtl8812AU_8821AU_linux-master/os_dep/linux/ioctl_cfg80211.c:5087:2: warning: (near initialization for ‘rtw_cfg80211_ops.mgmt_tx’)
    scripts/Makefile.build:257: recipe for target '/home/matthew/Downloads/rtl8812AU_8821AU_linux-master/os_dep/linux/ioctl_cfg80211.o' failed
    make[2]: *** [/home/matthew/Downloads/rtl8812AU_8821AU_linux-master/os_dep/linux/ioctl_cfg80211.o] Error 1
    Makefile:1394: recipe for target '_module_/home/matthew/Downloads/rtl8812AU_8821AU_linux-master' failed
    make[1]: *** [_module_/home/matthew/Downloads/rtl8812AU_8821AU_linux-master] Error 2
    make[1]: Leaving directory '/usr/src/linux-headers-3.19.0-15-generic'
    Makefile:1040: recipe for target 'modules' failed
    make: *** [modules] Error 2

bits3rpent, 2015-04-26:
I will help you, but this repo seems to mainly be inactive. Check out my repo and post the issue there https://github.com/bits3rpent/rtl8812AU_8821AU_linux

bits3rpent, 2015-04-26:
Download, build and install then if there is still issue post.

nzmm, 2015-04-26:
Worked a charm.  Thanks a bunch.

Matt

bits3rpent, 2015-04-26:
Glad it helped! Give it a star if you liked it :)

simonvizzini, 2015-04-26:
@bits3rpent: thanks for forking and updating this driver! I wanted to post this on your repo, but there is no issue tracker, thus I'll ask you here.

I'm using Arch Linux right now, and their current kernel is 3.19.3. I can compile the driver from your repo just fine under this kernel, but I just gave linux 4.0 a try and there it won't compile. Arch will probably switch soon to the 4.0 kernel. Do you have any idea what needs to be changed to make it compile on 4.0? Otherwise I'll be without wifi once 4.0 goes mainstream on Arch :( It's really sad to be dependent on this driver, and Realtek not caring at all for their product...

bits3rpent, 2015-04-26:
Hey Simon, I am going to create a new repo instead of a fork. This way, I can keep track of the issues. I will help you, and hopefully we can get it solved today! I will probably make an arch branch on my repo.

bits3rpent, 2015-04-26:
Never mind, I just added issues to my current repo. Please post there :)

simonvizzini, 2015-04-26:
Oh really nice, thank you! Let me know if there is anything I can do. For now I can give you the output of make :)

http://pastebin.com/1QWJQC5u

krichter722, 2015-04-28:
The issue seems to be a change of function arguments in `cfg80211_rx_mgmt`. My failing build log:

    /usr/bin/make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.0.0-040000-generic/build M=/mnt/richtercloud/sources/rtl8812AU_8821AU_linux  modules
    make[1]: Entering directory '/usr/src/linux-headers-4.0.0-040000-generic'
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/core/rtw_cmd.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/core/rtw_security.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/core/rtw_debug.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/core/rtw_io.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/core/rtw_ioctl_query.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/core/rtw_ioctl_set.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/core/rtw_ieee80211.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/core/rtw_mlme.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/core/rtw_mlme_ext.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/core/rtw_wlan_util.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/core/rtw_vht.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/core/rtw_pwrctrl.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/core/rtw_rf.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/core/rtw_recv.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/core/rtw_sta_mgt.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/core/rtw_ap.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/core/rtw_xmit.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/core/rtw_p2p.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/core/rtw_tdls.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/core/rtw_br_ext.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/core/rtw_iol.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/core/rtw_sreset.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/core/efuse/rtw_efuse.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/os_dep/osdep_service.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/os_dep/linux/os_intfs.o
    /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/os_dep/linux/os_intfs.c:1698:2: warning: initialization from incompatible pointer type
      .ndo_select_queue = rtw_select_queue,
      ^
    /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/os_dep/linux/os_intfs.c:1698:2: warning: (near initialization for ‘rtw_netdev_ops.ndo_select_queue’)
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/os_dep/linux/usb_intf.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/os_dep/linux/usb_ops_linux.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/os_dep/linux/ioctl_linux.o
    /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/os_dep/linux/ioctl_linux.c: In function ‘rtw_mp_efuse_get’:
    /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/os_dep/linux/ioctl_linux.c:8887:65: warning: iteration 16u invokes undefined behavior [-Waggressive-loop-optimizations]
         sprintf(extra, "%s%02X ", extra, pEfuseHal->fakeEfuseInitMap[i+j]);
                                                                     ^
    /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/os_dep/linux/ioctl_linux.c:8875:3: note: containing loop
       for (i = 0; i < EFUSE_MAP_SIZE; i += 16)
       ^
    /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/os_dep/linux/ioctl_linux.c:9307:69: warning: iteration 16u invokes undefined behavior [-Waggressive-loop-optimizations]
         sprintf(extra, "%s %02X", extra, pEfuseHal->fakeEfuseModifiedMap[i+j]);
                                                                         ^
    /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/os_dep/linux/ioctl_linux.c:9295:3: note: containing loop
       for (i=0; i<EFUSE_MAP_SIZE; i+=16)
       ^
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/os_dep/linux/xmit_linux.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/os_dep/linux/mlme_linux.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/os_dep/linux/recv_linux.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/os_dep/linux/ioctl_cfg80211.o
    /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/os_dep/linux/ioctl_cfg80211.c: In function ‘cfg80211_rtw_add_key’:
    /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/os_dep/linux/ioctl_cfg80211.c:1250:35: warning: passing argument 2 of ‘_rtw_memcpy’ discards ‘const’ qualifier from pointer target type
       _rtw_memcpy(param->u.crypt.seq, params->seq, params->seq_len);
                                       ^
    In file included from /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/include/drv_types.h:32:0,
                     from /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/os_dep/linux/ioctl_cfg80211.c:22:
    /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/include/osdep_service.h:167:13: note: expected ‘void *’ but argument is of type ‘const u8 *’
     extern void _rtw_memcpy(void* dec, void* sour, u32 sz);
                 ^
    /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/os_dep/linux/ioctl_cfg80211.c:1256:35: warning: passing argument 2 of ‘_rtw_memcpy’ discards ‘const’ qualifier from pointer target type
[... cut ...]
    scripts/Makefile.build:258: recipe for target '/mnt/richtercloud/sources/rtl8812AU_8821AU_linux/os_dep/linux/ioctl_cfg80211.o' failed
    make[2]: *** [/mnt/richtercloud/sources/rtl8812AU_8821AU_linux/os_dep/linux/ioctl_cfg80211.o] Error 1
    Makefile:1390: recipe for target '_module_/mnt/richtercloud/sources/rtl8812AU_8821AU_linux' failed
    make[1]: *** [_module_/mnt/richtercloud/sources/rtl8812AU_8821AU_linux] Error 2
    make[1]: Leaving directory '/usr/src/linux-headers-4.0.0-040000-generic'
    Makefile:1040: recipe for target 'modules' failed
    make: *** [modules] Error 2

TheBibliofilus, 2015-05-01:
@bits3rpent Thank you for this (had to sign up here to thank you!), it saved my day and my hair from being torn out and/or going gray.

I'm a Linux noob and have just switched/abandoned Windows as system OS and have been trying to set my my machine up and been wrestling with the Asus AC56 USB Wi-Fi-dongle, and now it magically works!

bits3rpent, 2015-05-01:
@TheBibliofilus I am so glad I could help my friend! If you like it, please star my repo so other people can see it more too :)

harshavardhana, 2015-05-04:
Should be fixed in the repo right now.

krichter722, 2015-05-04:
`STATION_INFO_SIGNAL`, `STATION_INFO_TX_BITRATE`, `STATION_INFO_RX_PACKETS`, `STATION_INFO_TX_PACKETS` and `STATION_INFO_ASSOC_REQ_IES` are still undeclared. The errors are the same like in the build log above (some errors and warnings are fixed though).

harshavardhana, 2015-05-04:
@krichter722 which version of kernel are you using? 

seems to be working fine on
~~~
Linux command 3.19.0-15-generic #15-Ubuntu SMP Thu Apr 16 23:32:37 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux
~~~ 

Will get it fixed..

krichter722, 2015-05-04:
4.0.0 and it fails with 4.0.1 as well. Thanks in advance.

bits3rpent, 2015-05-04:
I will work on 4.0 eventually. My priorities are 5ghz and performance issues right now.

harshavardhana, 2015-05-07:
I just pushed a change, can this be tested? @krichter722

closed 2015-05-07 by harshavardhana

krichter722, 2015-05-07:
in d5fd80bd70f76403b049a5f9cafb50eaec2ec172 `make clean && make install` fails with

    cd hal/OUTSRC/ ; rm -fr */*.mod.c */*.mod */*.o */.*.cmd */*.ko
    cd hal/OUTSRC/ ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko 
    cd hal/led ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
    cd hal ; rm -fr */*/*.mod.c */*/*.mod */*/*.o */*/.*.cmd */*/*.ko
    cd hal ; rm -fr */*.mod.c */*.mod */*.o */.*.cmd */*.ko
    cd hal ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
    cd core/efuse ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
    cd core ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
    cd os_dep/linux ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
    cd os_dep ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
    rm -fr Module.symvers ; rm -fr Module.markers ; rm -fr modules.order
    rm -fr *.mod.c *.mod *.o .*.cmd *.ko *~
    rm -fr .tmp_versions

    CLEAN SUCCESSFUL (total time: 669ms)
    /usr/bin/make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.0.1-040001-generic/build M=/mnt/richtercloud/sources/rtl8812AU_8821AU_linux  modules
    make[1]: Entering directory '/usr/src/linux-headers-4.0.1-040001-generic'
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/core/rtw_cmd.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/core/rtw_security.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/core/rtw_debug.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/core/rtw_io.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/core/rtw_ioctl_query.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/core/rtw_ioctl_set.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/core/rtw_ieee80211.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/core/rtw_mlme.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/core/rtw_mlme_ext.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/core/rtw_wlan_util.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/core/rtw_vht.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/core/rtw_pwrctrl.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/core/rtw_rf.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/core/rtw_recv.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/core/rtw_sta_mgt.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/core/rtw_ap.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/core/rtw_xmit.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/core/rtw_p2p.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/core/rtw_tdls.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/core/rtw_br_ext.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/core/rtw_iol.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/core/rtw_sreset.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/core/efuse/rtw_efuse.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/os_dep/osdep_service.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/os_dep/linux/os_intfs.o
    /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/os_dep/linux/os_intfs.c:1695:2: warning: initialization from incompatible pointer type
      .ndo_select_queue = rtw_select_queue,
      ^
    /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/os_dep/linux/os_intfs.c:1695:2: warning: (near initialization for ‘rtw_netdev_ops.ndo_select_queue’)
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/os_dep/linux/usb_intf.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/os_dep/linux/usb_ops_linux.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/os_dep/linux/ioctl_linux.o
    /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/os_dep/linux/ioctl_linux.c: In function ‘rtw_mp_efuse_get’:
    /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/os_dep/linux/ioctl_linux.c:8887:65: warning: iteration 16u invokes undefined behavior [-Waggressive-loop-optimizations]
         sprintf(extra, "%s%02X ", extra, pEfuseHal->fakeEfuseInitMap[i+j]);
                                                                     ^
    /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/os_dep/linux/ioctl_linux.c:8875:3: note: containing loop
       for (i = 0; i < EFUSE_MAP_SIZE; i += 16)
       ^
    /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/os_dep/linux/ioctl_linux.c:9307:69: warning: iteration 16u invokes undefined behavior [-Waggressive-loop-optimizations]
         sprintf(extra, "%s %02X", extra, pEfuseHal->fakeEfuseModifiedMap[i+j]);
                                                                         ^
    /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/os_dep/linux/ioctl_linux.c:9295:3: note: containing loop
       for (i=0; i<EFUSE_MAP_SIZE; i+=16)
       ^
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/os_dep/linux/xmit_linux.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/os_dep/linux/mlme_linux.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/os_dep/linux/recv_linux.o
      CC [M]  /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/os_dep/linux/ioctl_cfg80211.o
    /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/os_dep/linux/ioctl_cfg80211.c: In function ‘cfg80211_rtw_add_key’:
    /mnt/richtercloud/sources/rtl8812AU_8821AU_linux/os_dep/linux/ioctl_cfg80211.c:1250:35: warning: passing argument 2 of ‘_rtw_memcpy’ discards ‘const’ qualifier from pointer target type
[... cut ...]
    make[2]: *** [/mnt/richtercloud/sources/rtl8812AU_8821AU_linux/os_dep/linux/ioctl_cfg80211.o] Error 1
    Makefile:1390: recipe for target '_module_/mnt/richtercloud/sources/rtl8812AU_8821AU_linux' failed
    make[1]: *** [_module_/mnt/richtercloud/sources/rtl8812AU_8821AU_linux] Error 2
    make[1]: Leaving directory '/usr/src/linux-headers-4.0.1-040001-generic'
    Makefile:1041: recipe for target 'modules' failed
    make: *** [modules] Error 2

    BUILD FAILED (exit value 2, total time: 36s)

`BL80211_STA_INFO_TX_PACKETS` is still undefined.

harshavardhana, 2015-05-07:
> BL80211_STA_INFO_TX_PACKETS is still undefined.

oops typo *facepalm* , fixed try again.

krichter722, 2015-05-07:
It's better if you figure it out using #45.

commit d5fd80bd (2015-05-07): Support Linux 4.0.0, change enum NL82011 flags to match upstream changes - fixes #41

Also remove garbage characters from all files.