abperiasamy/rtl8812AU_8821AU_linux #42: Compiling for 3.10.72 results in "Unknown symbol"

tosiara, 2015-04-30:
Using latest master, insmod 8812ac give error:

[69856.895908@0] 8812au: Unknown symbol wiphy_new (err 0)
[69856.896040@0] 8812au: Unknown symbol cfg80211_del_sta (err 0)
[69856.896092@0] 8812au: Unknown symbol cfg80211_inform_bss_frame (err 0)

Checking out previous tree, like f1008f6d87e68a62807d46780827ff8aecfef8f0 resolves this

bits3rpent, 2015-05-01:
Please try my repo, this one is no longer updated. https://github.com/bits3rpent/rtl8812AU_8821AU_linux

Id2ndR, 2015-06-22:
bits3rpent's repo doesn't help. I managed to get it working by using https://github.com/gnab/rtl8812au repo and follow the ReadMe :
- just edit the Makefile with "CONFIG_PLATFORM_I386_PC = n" and "CONFIG_PLATFORM_ARM_RPI = y"

=> This will compile a module that does neither required wiphy nor cfg80211.

harshavardhana, 2015-06-22:
> bits3rpent's repo doesn't help. I managed to get it working by using https://github.com/gnab/rtl8812au repo and follow the ReadMe :

If you get a chance, can you send a pull request with this change?

Xenopathic, 2016-07-01:
I figured out why this happens: Hardkernel (makers of Odroid, which is the platform I'm assuming you're seeing these issues on) use a kernel with the 'backports' tree enabled, which includes newer networking components. In 3.10, this results in the kernel having a newer cfg80211 which exposes symbols such as `wiphy_new_nm()` rather than `wiphy_new()`. The required headers are there too, but they aren't used by the module compile system since they are in `/usr/src/linux-*/backports/include`, hence why this module gets compiled with the stock headers, mismatching the symbols in the kernel.

The solution is to get the kernel build system to check backports for include files. I can think of a really crude way by simply changing the kernel Makefile, but I don't know what the clean solution is (probably something needed upstream in the kernel itself).