abperiasamy/rtl8812AU_8821AU_linux #44: Does not compile (gcc: error: unrecognized command Werror=date-time)

Chris-Bee, 2015-05-07:
Hey, even if I remove this line (-Wdate-time) from the makefile I get further errors and the compiler says the Function "cfg80211_rx_mgmt" is not declared.

harshavardhana, 2015-05-07:
Can you let us know your kernel version?

Chris-Bee, 2015-05-07:
Of course and thanks for the fast response. The kernel version is 3.13.0-52-generic.

Thanks

harshavardhana, 2015-05-07:
Can you paste your error output perhaps like a compile error?  - it must be related to the kernel version.

Chris-Bee, 2015-05-07:
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/3.13.0-52-generic/build M=/rtl8812AU_8821AU_linux  modules
make[1]: Entering directory `/usr/src/linux-headers-3.13.0-52-generic'
  CC [M]  /rtl8812AU_8821AU_linux/core/rtw_cmd.o
cc1: error: -Werror=date-time: no option -Wdate-time
make[2]: *** [/rtl8812AU_8821AU_linux/core/rtw_cmd.o] Error 1
make[1]: *** [_module_/rtl8812AU_8821AU_linux] Error 2
make[1]: Leaving directory `/usr/src/linux-headers-3.13.0-52-generic'
make: *** [modules] Error 2

krichter722, 2015-05-07:
The project seems to compile with `gcc` >4.9 only. I didn't manage to do it with 4.8 (failing with the same error). Maybe it's time for `autoconf`?!

harshavardhana, 2015-05-08:
> Maybe it's time for autoconf?!

autoconf is a bad idea for kernel builds, in-fact i need to figure out why that option even is used here.

commit 12217798 (2015-05-08): date-time warning suppression is not needed fixes #44
closed 2015-05-08 by harshavardhana