abperiasamy/rtl8812AU_8821AU_linux #47: Errors on "cfg80211_rx_mgmt"

Chris-Bee, 2015-05-08:
During the compilation I get some Errors related to the function cfg80211_rx_mgmt:

make[1]: Entering directory `/usr/src/linux-headers-3.13.0-52-generic'

rtl8812AU_8821AU_linux/os_dep/linux/ioctl_cfg80211.c:3897:2: error: too few arguments to function ‘cfg80211_rx_mgmt’
  rtw_cfg80211_rx_mgmt(padapter, freq, 0, pmgmt_frame, frame_len, GFP_ATOMIC);

rtl8812AU_8821AU_linux/os_dep/linux/ioctl_cfg80211.c:22:
include/net/cfg80211.h:4102:6: note: declared here
 bool cfg80211_rx_mgmt(struct wireless_dev *wdev, int freq, int sig_dbm,

rtl8812AU_8821AU_linux/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_rx_action’:
rtl8812AU_8821AU_linux/os_dep/linux/ioctl_cfg80211.c:3965:2: error: too few arguments to function ‘cfg80211_rx_mgmt’

include/net/cfg80211.h:4102:6: note: declared here
 bool cfg80211_rx_mgmt(struct wireless_dev *wdev, int freq, int sig_dbm,

harshavardhana, 2015-05-08:
Looks like the kernel changes have broken this function ``cfg80211_rx_mgmt`` , all our test was on 14.04 and above

harshavardhana, 2015-05-08:
Will see how i can get it fixed for 3.13.0 kernel

harshavardhana, 2015-05-08:
Can you test this? 

Inside git clone of this project
~~~
# git fetch origin 
# git checkout 3.13
# make
~~~

Chris-Bee, 2015-05-08:
That works fine, perfect :+1:

closed 2015-05-10 by harshavardhana