abperiasamy/rtl8812AU_8821AU_linux #49: Compilation in Arch Linux Kernel Version 3.4.103

nagarjunr, 2015-05-13:
I am trying to cross-compile the Edimax EW7811UTC driver for Arch Linux-3.4.103 in Ubuntu. But I am always getting the following message without *.ko file. 

COMMAND:
make ARCH=arm CROSS_COMPILE=${CC201310} -C /home/nagarjun/linux-sunxi/cubie/sunx7i_22/ M=$PWD

Where,
${CC201310} : is the cross compiler path, which was set properly
$PWD : is the driver directory
/home/nagarjun/linux-sunxi/cubie/sunx7i_22 : My Arch Linux Kernel Source

OUTPUT:
make: Entering directory `/home/nagarjun/linux-sunxi/cubie/sunx7i_22'
  LD      /home/nagarjun/Desktop/EW7811UTC/built-in.o
  Building modules, stage 2.
  MODPOST 0 modules
make: Leaving directory `/home/nagarjun/linux-sunxi/cubie/sunx7i_22'

simonsmiley, 2015-05-14:
it's not working for me either. But why are you using a 3.4 kernel with archlinux?! that's a couple years old now... even the lts-kernel is 3.14

nagarjunr, 2015-05-15:
Its because my hardware(cubieboard 2) recommends 3.4 as the stable kernel version.

harshavardhana, 2015-05-21:
Have you tried

~~~
make ARCH=arm CROSS_COMPILE=${CC201310} -C /home/nagarjun/linux-sunxi/cubie/sunx7i_22/ M=$PWD modules
~~~ 

And see if it executes your compiler?

harshavardhana, 2015-05-21:
I think the problem is in how make is invoking these parameters internally. I would suggest to do this instead

~~~
# export CROSS_COMPILE=$(CC201310)
# export ARCH=arm
# export KSRC="/home/nagarjun/linux-sunxi/cubie/sunx7i_22/"

# make
~~~

do not need to set M it automatically uses PWD as your module directory

nagarjunr, 2015-05-22:
I tried both the methods. It wasn't working. Atlast I was able to compile it successfully. It was due to errors in Makefile and the incompatibility between kernel source code and the driver source. After minor fixes in Makefile & Kernel Source, it was compiled successfully. Thanks for the information.

closed 2015-05-22 by nagarjunr