abperiasamy/rtl8812AU_8821AU_linux #50: Linux 4

bartonc, 2015-05-14:
Hi,

I was wondering if you are going to update the drivers so that they are able to build on linux kernel version 4.

Tried building on both of my linux boxes got the following error:

 make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.0.2-gentoo/build M=/root/wireless_drv/k4/rtl8812AU_8821AU_linux-master modules make[1]: Entering directory '/usr/src/linux-4.0.2-gentoo' CC [M] /root/wireless_drv/k4/rtl8812AU_8821AU_linux-master/os_dep/linux/ioctl_cfg80211.o scripts/Makefile.build:258: recipe for target '/root/wireless_drv/k4/rtl8812AU_8821AU_linux-master/os_dep/linux/ioctl_cfg80211.o' failed Makefile:1390: recipe for target '_module_/root/wireless_drv/k4/rtl8812AU_8821AU_linux-master' failed make[1]: Leaving directory '/usr/src/linux-4.0.2-gentoo' Makefile:1040: recipe for target 'modules' failed

simonsmiley, 2015-05-14:
failed for me too... here is my make log
http://sprunge.us/JJde
tried using 4.0.0 and 4.0.1 running archlinux

harshavardhana, 2015-05-15:
> http://sprunge.us/JJde

cannot access this link - do you have a better one?

simonsmiley, 2015-05-16:
seems like sprunge is having some issues... using pastebin instead
http://pastebin.com/8TQFDm1H

simonsmiley, 2015-05-19:
seems like sprunge is back up agian. The logs are actually from two different builds using a PKGBUILD from aur.archlinux.org and compiling it by hand with the instructions provided in github

simonsmiley, 2015-05-19:
I managed to build it using this repository:
https://github.com/csssuf/rtl8812au
(I really don't like the fragmentation for rtl8812au on github, there are dozens of repos...)
running kernel version 4.0.3 now.
will have to see if everything is working when I get home since I don't have the dongle with me right now...

bartonc, 2015-05-19:
Thanks for the update, updated my kernel on arch as well and the drivers installed for me as well.

simonsmiley, 2015-05-19:
did you check the gentoo wiki about that, they have an article about this driver. I got the repository from there...
@harshavardhana is there any chance of merging the two projects? I didn't check the differences yet. I guess it's in everybody's best interest to have one repository that works for everyone and not split it up...

harshavardhana, 2015-05-19:
> @harshavardhana is there any chance of merging the two projects? I didn't check the differences yet. 
> I guess it's in everybody's best interest to have one repository that works for everyone and not split it
> up...

I will write to them over this week and see if we can work anything out to move this into a collaboration.

bartonc, 2015-05-23:
Okay the patches resolved the issue for my gentoo box.

Thank-you

closed 2015-05-23 by bartonc