abperiasamy/rtl8812AU_8821AU_linux #51: Bad connection quality

sambarluc, 2015-05-17:
Using Ubuntu kernel 3.19.0-16-generic, I can compile and install without any issue. However, signal quality is very bad:

wlan1     Scan completed :
                    Protocol:IEEE 802.11bgn
                    Mode:Master
                    Frequency:2.452 GHz (Channel 9)
                    Encryption key:on
                    Bit Rates:144 Mb/s
                    IE: WPA Version 1
                        Group Cipher : TKIP
                        Pairwise Ciphers (1) : TKIP
                        Authentication Suites (1) : PSK
                    IE: IEEE 802.11i/WPA2 Version 1
                        Group Cipher : TKIP
                        Pairwise Ciphers (1) : CCMP
                        Authentication Suites (1) : PSK
                    Quality=0/100  Signal level=23/100  

On a different linux machine (Mint kernel 3.13.0-24) I can use the standard driver from Alfa web site, and I get very good connection quality.

harshavardhana, 2015-05-17:
Driver needs an update, certain cards are known to have issues.  Best performant external card i have found is EDIMAX AC1200

sambarluc, 2015-05-17:
Well, I would like to use this card, it's the one I have. Can I hope in a fix any time soon?

harshavardhana, 2015-05-18:
> Can I hope in a fix any time soon?

There is no way for us to test it since we do not carry the same hardware. If you can help in debugging then we might to start looking at the debugging output, any kernel traces - dmidecode output etc.

You should use the standard driver from Alfa website if that works properly.

joseph-jja, 2015-07-16:
You are connecting to the 2.4Ghz band.  Probably the 20Mhz frequency, so 144Mbps is the fastest you get on that band and frequency.  Change the frequency to 40Mhz and you can get 300Mbps. You'll need a 5Ghz access point to get 5Ghz speeds.

closed 2019-04-01 by sambarluc