abperiasamy/rtl8812AU_8821AU_linux #53: ioctl_cfg80211.c-kernel_4.0.4.patch

hackdefendr, 2015-05-21:
Minor patch to os_dep/linux/ioctl_cfg80211.c allowing module to compile under Kernel 4.0.4 (technically this should work with >= 4.0.0.

Test on Kali Linux w custom Kernel v4.0.4 using 'make oldconfig'.

DMESG:
```
[ 5830.174485] usbcore: registered new interface driver rtl8812au
```

PATCH:
```
--- os_dep/linux/ioctl_cfg80211.c.orig	2015-05-21 00:37:26.477166369 -0500
+++ os_dep/linux/ioctl_cfg80211.c	2015-05-21 00:44:07.677154960 -0500
@@ -1422,31 +1422,33 @@
 			goto exit;
 		}
 
-#if LINUX_VERSION_CODE >= KERNEL_VERSION(4,0,0)
-		sinfo->filled |= BIT(NL80211_STA_INFO_SIGNAL);
-		sinfo->signal = translate_percentage_to_dbm(padapter->recvpriv.signal_strength);
-
-		sinfo->filled |= BIT(NL80211_STA_INFO_TX_BITRATE);
-		sinfo->txrate.legacy = rtw_get_cur_max_rate(padapter);
-
-		sinfo->filled |= BIT(NL80211_STA_INFO_RX_PACKETS);
-		sinfo->rx_packets = sta_rx_data_pkts(psta);
-
-		sinfo->filled |= BIT(NL80211_STA_INFO_TX_PACKETS);
-		sinfo->tx_packets = psta->sta_stats.tx_pkts;
+#if (LINUX_VERSION_CODE >= KERNEL_VERSION(4,0,0))
+       sinfo->filled |= BIT(NL80211_STA_INFO_SIGNAL);
 #else
-                sinfo->filled |= STATION_INFO_SIGNAL;
-		sinfo->signal = translate_percentage_to_dbm(padapter->recvpriv.signal_strength);
+        sinfo->filled |= STATION_INFO_SIGNAL;
+#endif
+        sinfo->signal = translate_percentage_to_dbm(padapter->recvpriv.signal_strength);
 
-		sinfo->filled |= STATION_INFO_TX_BITRATE;
-		sinfo->txrate.legacy = rtw_get_cur_max_rate(padapter);
+#if (LINUX_VERSION_CODE >= KERNEL_VERSION(4,0,0))
+       sinfo->filled |= BIT(NL80211_STA_INFO_TX_BITRATE);
+#else
+        sinfo->filled |= STATION_INFO_TX_BITRATE;
+#endif
+        sinfo->txrate.legacy = rtw_get_cur_max_rate(padapter);
 
-		sinfo->filled |= STATION_INFO_RX_PACKETS;
-		sinfo->rx_packets = sta_rx_data_pkts(psta);
+#if (LINUX_VERSION_CODE >= KERNEL_VERSION(4,0,0))
+       sinfo->filled |= BIT(NL80211_STA_INFO_RX_PACKETS);
+#else
+        sinfo->filled |= STATION_INFO_RX_PACKETS;
+#endif
+        sinfo->rx_packets = sta_rx_data_pkts(psta);
 
-		sinfo->filled |= STATION_INFO_TX_PACKETS;
-		sinfo->tx_packets = psta->sta_stats.tx_pkts;
+#if (LINUX_VERSION_CODE >= KERNEL_VERSION(4,0,0))
+       sinfo->filled |= BIT(NL80211_STA_INFO_TX_PACKETS);
+#else
+        sinfo->filled |= STATION_INFO_TX_PACKETS;
 #endif
+        sinfo->tx_packets = psta->sta_stats.tx_pkts;
 
 	}
 
@@ -3032,6 +3034,10 @@
 		else // WIFI_REASSOCREQ
 			ie_offset = _REASOCREQ_IE_OFFSET_;
 
+        sinfo.filled = 0;
+#if (LINUX_VERSION_CODE < KERNEL_VERSION(4,0,0))
+        sinfo.filled = STATION_INFO_ASSOC_REQ_IES;
+#endif
 		sinfo.assoc_req_ies = pmgmt_frame + WLAN_HDR_A3_LEN + ie_offset;
 		sinfo.assoc_req_ies_len = frame_len - WLAN_HDR_A3_LEN - ie_offset;
 		cfg80211_new_sta(ndev, GetAddr2Ptr(pmgmt_frame), &sinfo, GFP_ATOMIC);
```

harshavardhana, 2015-05-21:
Please make a pull request , what is the difference between this and current master? - i see that the current already compiles fine on  > 4.0.0  ?

hackdefendr, 2015-05-21:
It did compile ... but dmesg showed this error when loading:

```
[ 2202.749220] 8812au: Unknown symbol strnicmp (err 0)
```

harshavardhana, 2015-05-21:
> [ 2202.749220] 8812au: Unknown symbol strnicmp (err 0)

Ah understood, can you send a pull request for #54  ?

harshavardhana, 2015-05-28:
Were you able to see any issues here?

closed 2016-09-11 by harshavardhana