abperiasamy/rtl8812AU_8821AU_linux #54: rtw_android.c-kernel_4.0.4.patch

hackdefendr, 2015-05-21:
Minor patch to os_dep/linux/ioctl_cfg80211.c allowing module to compile under Kernel 4.0.4.

Tested on Kali Linux with customer Kernel v4.0.4 using 'make oldconfig'.

NOTE: Should be used with ioctl_cfg80211.c-kernel_4.0.4.patch Issue #53

PATCH:
```
--- os_dep/linux/rtw_android.c.orig	2015-05-21 00:44:41.001154012 -0500
+++ os_dep/linux/rtw_android.c	2015-05-21 00:48:18.933147814 -0500
@@ -224,10 +224,14 @@
 {
 	int cmd_num;
 	for(cmd_num=0 ; cmd_num<ANDROID_WIFI_CMD_MAX; cmd_num++)
-		if(0 == strnicmp(cmdstr , android_wifi_cmd_str[cmd_num], strlen(android_wifi_cmd_str[cmd_num])) )
-			break;
-		
-	return cmd_num;
+#if (LINUX_VERSION_CODE >= KERNEL_VERSION(4,0,0))
+        if(0 == strncasecmp(cmdstr , android_wifi_cmd_str[cmd_num], strlen(android_wifi_cmd_str[cmd_num])) )
+#else
+        if(0 == strnicmp(cmdstr , android_wifi_cmd_str[cmd_num], strlen(android_wifi_cmd_str[cmd_num])) )
+#endif
+            break;
+
+    return cmd_num;
 }
 
 int rtw_android_get_rssi(struct net_device *net, char *command, int total_len)
```

harshavardhana, 2015-05-21:
LGTM.

hackdefendr, 2015-05-21:
Additional dmesg after plugging in my AC1200-USB-AC56

```
[ 5830.174485] usbcore: registered new interface driver rtl8812au
[ 7076.664531] usb 2-1: new high-speed USB device number 2 using xhci_hcd
[ 7076.793007] usb 2-1: New USB device found, idVendor=0b05, idProduct=17d2
[ 7076.793011] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 7076.793013] usb 2-1: Product: 802.11n NIC
[ 7076.793014] usb 2-1: Manufacturer: Realtek
[ 7076.793016] usb 2-1: SerialNumber: 123456
[ 7077.533194] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
```

hackdefendr, 2015-05-21:
Pull request created.  I will work on #53 shortly.

harshavardhana, 2015-05-21:
Closed at #55

closed 2015-05-21 by harshavardhana