abperiasamy/rtl8812AU_8821AU_linux #63: Linux kernel 4.0.5 regression

mar77i, 2015-06-10:
Both rtl8187 drivers from here and from [0] don't work with the D-Link wifi adapter 2001:3314. The devices are registered with xhci and mtp tests (which fail in 4.0.4 too), but no device will appear. It does work in linux 4.0.4 but I don't know the codebase enough to investigate myself.

My distro usually ships vanilla packages, please let me know whether you can reproduce the issue.

[0] https://github.com/csssuf/rtl8812au

harshavardhana, 2015-06-22:
> My distro usually ships vanilla packages, please let me know whether you can reproduce the issue.

Do not carry the device myself,  it could be API level changes which keep occurring on Linux on a regular basis. There has been a large ifdef to handle that looks like 4.x would bring in more. 

Hard to track them from here, if there is no particular reason to use 4.0.5 kernel why not just use a working kernel instead?

mar77i, 2015-06-23:
I ended up putting the kernel on noupdate at some point as soon as I'd thought of it. I might follow up with bisecting Greg's kernel.org repo whenever I get to do that - in case that's the thing I can do for you.