abperiasamy/rtl8812AU_8821AU_linux #64: compile warnings on 3.13

joseph-jja, 2015-06-12:
In os_dep/linux/os_intfs.c 

if you change

static u16 rtw_select_queue(struct net_device *dev, struct sk_buff *skb)

static u16 rtw_select_queue(struct net_device *dev, struct sk_buff *skb, void *accel_priv, select_queue_fallback_t fallback)

It gets rid of a few compiler warnings.  It's been a while since I did C so I forget what happens when you call a function with wrong number of arguments.

I'm also wondering why it compiles os_dep/linux/rtw_android.c ? on x86?

harshavardhana, 2015-06-12:
Should be fixed.. in current master following fixes from @AndyPi

joseph-jja, 2015-06-19:
harshavardhana thanks, so which repo is the main repo and who's maintaining it?

harshavardhana, 2015-06-19:
> harshavardhana thanks, so which repo is the main repo and who's maintaining it?

Currently i am maintaining this.. so if you 'git clone' this master branch should have these fixed.

joseph-jja, 2015-07-18:
Works

closed 2015-07-18 by joseph-jja