abperiasamy/rtl8812AU_8821AU_linux #68: installed and loaded the module but adapter is not found in iwconfig

eugenio, 2015-07-09:
I successfully compiled and installed the driver but my adapter doesn't show up in iwconfig : 
>eth0      no wireless extensions.
>
>ppp0      no wireless extensions.
>
>lo        no wireless extensions.

dmesg shows: 

>[ 3562.933587] usbcore: registered new interface driver rtl8812au
>[ 3679.345829] usb 2-1.3: USB disconnect, device number 9
>[ 3695.437672] usb 2-1.2: new high-speed USB device number 10 using ehci-pci
>[ 3695.530517] usb 2-1.2: New USB device found, idVendor=0411, idProduct=025d
>[ 3695.530520] usb 2-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
>[ 3695.530522] usb 2-1.2: Product: WI-U3-866D
>[ 3695.530523] usb 2-1.2: Manufacturer: Buffalo
>[ 3695.530524] usb 2-1.2: SerialNumber: 123456

when i insert the dongle

but 

>modinfo 8812au 

doesn't list my Product ID, my system is debian jessie with latest patches. Can you help me?

harshavardhana, 2015-07-10:
> doesn't list my Product ID, my system is debian jessie with latest patches. Can you help me?

We might have to add your product id for it get registered with the driver, can you provide more info perhaps "lsusb -v" ?

eugenio, 2015-07-10:
Vendor id and product id are listed below:
> idVendor=0411, idProduct=025d
at the moment i don't have the device with me but i managed to have it active by adding : 
>#ifdef CONFIG_RTL8812A
>        {USB_DEVICE(0x0411, 0x025D),.driver_info = RTL8812}, /* BUFFALO */

in 

>usb_intf.c

and recompiling, but i have yet to find a working configuration with hostapd, moreover the driver seems to be a little unstable, with hostapd complaining about not being able to communicate with the driver or something.  other info i can provide later than 17:00 JST , sorry.