abperiasamy/rtl8812AU_8821AU_linux #69: Rasp 2 B w/ 4.0.7 works - iw station dump doesn't work

humberg, 2015-07-09:
Hi,

on my Rasp 2 with linux-headers 4.0.7, I compile and running with TP-Link T4U
It works!!!!
Very quick!
ac Mode!

:-)

But .... when I use the 8812au.ko, the  "iw dev wlan0 station dump" gives me nothing  :-(
No information with this command - when I change the Stick / Realtek or so, it works.

I forget to turn on some parameters in Makefile ????

Best regards

Markus

harshavardhana, 2015-07-10:
> I forget to turn on some parameters in Makefile ????

unsure, no way for us to test though no hardware.. Any debugging messages you can share? compile with DEBUG option perhaps

joseph-jja, 2015-07-15:
I'm on a pi and I get the same thing from station dump :( below shows nothing

sudo iw dev wlan1 station dump

how do you compile with debug?