abperiasamy/rtl8812AU_8821AU_linux #70: Truncated IP packets '4 bytes missing!'

praveenbhamidipati, 2015-07-11:
I have purchased D-Link DWA-171. I pulled down the sources from master branch and found that my card DWA-171 _(`13d3:3394`)_ was not represented in `os_dep/linux/usb_intf.c`. I added an entry just above DWA-172 corresponding to the USB vendor and device ids of my card, built, installed and restarted to get a wireless interface corresponding to DWA-171.
```
wlxec22446688AA  IEEE 802.11AC  ESSID:"AbcWiFi"  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency:5.18 GHz  Access Point: 00:11:33:55:77:99   
          Bit Rate:87 Mb/s   Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality=100/100  Signal level=100/100  Noise level=0/100
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
```

My issue is that 802.11n works just fine, but 802.11 ac has issues when accessing some websites _(i am writing this ticket using 802.11ac)_. The following is what I observed in `tcpdump`:
* 802.11n _(tcp traffic captured for command: `curl 'https://accounts.google.com/ServiceLogin'`)_
```
23:02:55.749480 IP 192.168.254.196.47797 > 216.58.216.173.443: Flags [S], seq 1355553534, win 29200, options [mss 1460,sackOK,TS val 1802390 ecr 0,nop,wscale 7], length 0
23:02:55.770408 IP 216.58.216.173.443 > 192.168.254.196.47797: Flags [S.], seq 857123418, ack 1355553535, win 42540, options [mss 1430,sackOK,TS val 922849898 ecr 1802390,nop,wscale 7], length 0
23:02:55.770510 IP 192.168.254.196.47797 > 216.58.216.173.443: Flags [.], ack 1, win 229, options [nop,nop,TS val 1802396 ecr 922849898], length 0
23:02:55.939580 IP 192.168.254.196.47797 > 216.58.216.173.443: Flags [P.], seq 1:288, ack 1, win 229, options [nop,nop,TS val 1802438 ecr 922849898], length 287
23:02:55.951147 IP 216.58.216.173.443 > 192.168.254.196.47797: Flags [.], ack 288, win 341, options [nop,nop,TS val 922850088 ecr 1802438], length 0
23:02:55.957441 IP 216.58.216.173.443 > 192.168.254.196.47797: Flags [.], seq 1:1419, ack 288, win 341, options [nop,nop,TS val 922850089 ecr 1802438], length 1418
23:02:55.957518 IP 192.168.254.196.47797 > 216.58.216.173.443: Flags [.], ack 1419, win 251, options [nop,nop,TS val 1802442 ecr 922850089], length 0
23:02:55.957442 IP 216.58.216.173.443 > 192.168.254.196.47797: Flags [.], seq 1419:2837, ack 288, win 341, options [nop,nop,TS val 922850089 ecr 1802438], length 1418
23:02:55.957587 IP 192.168.254.196.47797 > 216.58.216.173.443: Flags [.], ack 2837, win 274, options [nop,nop,TS val 1802442 ecr 922850089], length 0
23:02:55.957443 IP 216.58.216.173.443 > 192.168.254.196.47797: Flags [P.], seq 2837:3542, ack 288, win 341, options [nop,nop,TS val 922850089 ecr 1802438], length 705
23:02:55.957609 IP 192.168.254.196.47797 > 216.58.216.173.443: Flags [.], ack 3542, win 296, options [nop,nop,TS val 1802442 ecr 922850089], length 0
23:02:55.961625 IP 192.168.254.196.47797 > 216.58.216.173.443: Flags [P.], seq 288:363, ack 3542, win 296, options [nop,nop,TS val 1802443 ecr 922850089], length 75
23:02:56.013087 IP 216.58.216.173.443 > 192.168.254.196.47797: Flags [.], ack 363, win 341, options [nop,nop,TS val 922850150 ecr 1802443], length 0
23:02:56.013157 IP 192.168.254.196.47797 > 216.58.216.173.443: Flags [P.], seq 363:414, ack 3542, win 296, options [nop,nop,TS val 1802456 ecr 922850150], length 51
23:02:56.025003 IP 216.58.216.173.443 > 192.168.254.196.47797: Flags [.], ack 414, win 341, options [nop,nop,TS val 922850161 ecr 1802456], length 0
23:02:56.025009 IP 216.58.216.173.443 > 192.168.254.196.47797: Flags [P.], seq 3542:3788, ack 414, win 341, options [nop,nop,TS val 922850161 ecr 1802456], length 246
23:02:56.028251 IP 192.168.254.196.47797 > 216.58.216.173.443: Flags [P.], seq 414:526, ack 3788, win 318, options [nop,nop,TS val 1802460 ecr 922850161], length 112
23:02:56.050160 IP 216.58.216.173.443 > 192.168.254.196.47797: Flags [P.], seq 3788:4500, ack 526, win 341, options [nop,nop,TS val 922850187 ecr 1802460], length 712
23:02:56.050975 IP 192.168.254.196.47797 > 216.58.216.173.443: Flags [P.], seq 526:557, ack 4500, win 340, options [nop,nop,TS val 1802466 ecr 922850187], length 31
23:02:56.061021 IP 216.58.216.173.443 > 192.168.254.196.47797: Flags [F.], seq 4500, ack 557, win 341, options [nop,nop,TS val 922850199 ecr 1802466], length 0
23:02:56.077640 IP 192.168.254.196.47797 > 216.58.216.173.443: Flags [F.], seq 557, ack 4501, win 340, options [nop,nop,TS val 1802472 ecr 922850199], length 0
23:02:56.089389 IP 216.58.216.173.443 > 192.168.254.196.47797: Flags [.], ack 558, win 341, options [nop,nop,TS val 922850226 ecr 1802472], length 0
^C
22 packets captured
22 packets received by filter
0 packets dropped by kernel
```
* 802.11ac _(tcp traffic captured for command: `curl 'https://accounts.google.com/ServiceLogin'`)_
```
23:03:43.491502 IP 192.168.254.196.47798 > 216.58.216.173.443: Flags [S], seq 690414722, win 29200, options [mss 1460,sackOK,TS val 1814326 ecr 0,nop,wscale 7], length 0
23:03:43.498775 IP 216.58.216.173.443 > 192.168.254.196.47798: Flags [S.], seq 2486729204, ack 690414723, win 42540, options [mss 1430,sackOK,TS val 922887422 ecr 1814326,nop,wscale 7], length 0
23:03:43.498878 IP 192.168.254.196.47798 > 216.58.216.173.443: Flags [.], ack 1, win 229, options [nop,nop,TS val 1814328 ecr 922887422], length 0
23:03:43.664859 IP 192.168.254.196.47798 > 216.58.216.173.443: Flags [P.], seq 1:288, ack 1, win 229, options [nop,nop,TS val 1814369 ecr 922887422], length 287
23:03:43.673688 IP 216.58.216.173.443 > 192.168.254.196.47798: Flags [.], ack 288, win 341, options [nop,nop,TS val 922887595 ecr 1814369], length 0
23:03:43.673696 IP 216.58.216.173.443 > 192.168.254.196.47798: Flags [.], seq 1:1419, ack 288, win 341, options [nop,nop,TS val 922887596 ecr 1814369], length 1418
23:03:43.673794 IP 192.168.254.196.47798 > 216.58.216.173.443: Flags [.], ack 1419, win 251, options [nop,nop,TS val 1814371 ecr 922887596], length 0
23:03:43.673704 IP 216.58.216.173.443 > 192.168.254.196.47798: Flags [.], seq 1419:2837, ack 288, win 341, options [nop,nop,TS val 922887596 ecr 1814369], length 1418
23:03:43.673855 IP 192.168.254.196.47798 > 216.58.216.173.443: Flags [.], ack 2837, win 274, options [nop,nop,TS val 1814371 ecr 922887596], length 0
23:03:43.673710 IP 216.58.216.173.443 > 192.168.254.196.47798: Flags [P.], seq 2837:3542, ack 288, win 341, options [nop,nop,TS val 922887596 ecr 1814369], length 705
23:03:43.673880 IP 192.168.254.196.47798 > 216.58.216.173.443: Flags [.], ack 3542, win 296, options [nop,nop,TS val 1814371 ecr 922887596], length 0
23:03:43.677563 IP 192.168.254.196.47798 > 216.58.216.173.443: Flags [P.], seq 288:363, ack 3542, win 296, options [nop,nop,TS val 1814372 ecr 922887596], length 75
23:03:43.752635 IP 216.58.216.173.443 > 192.168.254.196.47798: Flags [.], ack 363, win 341, options [nop,nop,TS val 922887676 ecr 1814372], length 0
23:03:43.752708 IP 192.168.254.196.47798 > 216.58.216.173.443: Flags [P.], seq 363:414, ack 3542, win 296, options [nop,nop,TS val 1814391 ecr 922887676], length 51
23:03:43.759164 IP 216.58.216.173.443 > 192.168.254.196.47798: Flags [.], ack 414, win 341, options [nop,nop,TS val 922887683 ecr 1814391], length 0
23:03:43.759182 IP truncated-ip - 4 bytes missing! 216.58.216.173.443 > 192.168.254.196.47798: Flags [P.], seq 3542:3788, ack 414, win 341, options [nop,nop,TS val 922887683 ecr 1814391], length 246
23:03:43.975616 IP truncated-ip - 4 bytes missing! 216.58.216.173.443 > 192.168.254.196.47798: Flags [P.], seq 3542:3788, ack 414, win 341, options [nop,nop,TS val 922887900 ecr 1814391], length 246
23:03:44.195875 IP truncated-ip - 4 bytes missing! 216.58.216.173.443 > 192.168.254.196.47798: Flags [P.], seq 3542:3788, ack 414, win 341, options [nop,nop,TS val 922888117 ecr 1814391], length 246
23:03:44.633093 IP truncated-ip - 4 bytes missing! 216.58.216.173.443 > 192.168.254.196.47798: Flags [P.], seq 3542:3788, ack 414, win 341, options [nop,nop,TS val 922888551 ecr 1814391], length 246
23:03:45.500508 IP truncated-ip - 4 bytes missing! 216.58.216.173.443 > 192.168.254.196.47798: Flags [P.], seq 3542:3788, ack 414, win 341, options [nop,nop,TS val 922889419 ecr 1814391], length 246
23:03:47.235087 IP truncated-ip - 4 bytes missing! 216.58.216.173.443 > 192.168.254.196.47798: Flags [P.], seq 3542:3788, ack 414, win 341, options [nop,nop,TS val 922891155 ecr 1814391], length 246
[... cut ...]
          Usage Type               Data
        wMaxPacketSize     0x0031  1x 49 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            1
          Transfer Type            Isochronous
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0031  1x 49 bytes
        bInterval               1
```

andrew2412, 2015-08-18:
I have the same issue. I noticed that each packet with length 41, 42, 666, 977 and 1133 bytes are incorrectly receive by adapter. In network N everything goes OK. The problem is with AC network.

mayli, 2018-02-15:
having same issue on my `4.9.35+ #1014 Fri Jun 30 14:34:49 BST 2017 armv6l GNU/Linux` and `0bda:a811`

this is blocking some DHCP replies

```
00:23:13.081519 IP truncated-ip - 4 bytes missing! (tos 0x0, ttl 64, id 0, offset 0, flags [DF], proto UDP (17), length 360)
    10.x.x.x.bootps > 10.x.x.x.x.bootpc: [no cksum] BOOTP/DHCP, Reply, length 332, xid 0xd581e0bf, Flags [none] (0x0000)
```