abperiasamy/rtl8812AU_8821AU_linux #75: Raspberry Pi2 Installation problems (Raspbian)

Jasimo, 2015-07-20:
Hi,

I just want to ask if somebody can support me.
I tried to install the driver under Raspbian 3.18.11-v7+ #781, I followed the steps discribed here:
```sudo apt-get install linux-image-rpi-rpfv linux-headers-rpi-rpfv dkms build-essential bc```
ends up with the message:
```ln: failed to create hard link `/boot/initrd.img-3.18.0-trunk-rpi.dpkg-bak' => `/boot/initrd.img-3.18.0-trunk-rpi': Operation not permitted
update-initramfs: Generating /boot/initrd.img-3.18.0-trunk-rpi```
if I ignore this and go further entering
```
kernel=vmlinuz-3.10-3-rpi
initramfs initrd.img-3.10-3-rpi followkernel
```
to /boot/config.txt, my Pi2 doesn´t boot anymore.  :disappointed_relieved:
Can anyone help me on this?
rgs
Jan

harshavardhana, 2015-07-20:
> to /boot/config.txt, my Pi2 doesn´t boot anymore.

do you happen to remember the kernel version which used to boot?

Jasimo, 2015-07-20:
it was 3.18.11-v7+

finally i got it to work now, the problem was the missing kernel headers for 3.18.11-v7+
I get around this issue by using rpi-source
http://github.com/notro/rpi-source/wiki
for getting the right kernel headers, after this I was able to compile and load the kernel module successfully

harshavardhana, 2015-07-20:
> for getting the right kernel headers, after this I was able to compile and load the kernel module successfully

good to know.

closed 2015-07-20 by harshavardhana

maxyclu, 2018-01-04:
Hi,

I ran into the same problem with kernel version (a raspbian run on a raspberry pi 3):
Linux rpi3 4.9.35-v7+ #1014 SMP Fri Jun 30 14:47:43 BST 2017 armv7l GNU/Linux

And the error messages are:
Processing triggers for initramfs-tools (0.120+deb8u3) ...
ln: failed to create hard link ‘/boot/initrd.img-4.9.0-4-rpi.dpkg-bak’ => ‘/boot/initrd.img-4.9.0-4-rpi’: Operation not permitted
cp: error writing ‘/boot/initrd.img-4.9.0-4-rpi.dpkg-bak’: No space left on device
cp: failed to extend ‘/boot/initrd.img-4.9.0-4-rpi.dpkg-bak’: No space left on device
dpkg: error processing package initramfs-tools (--configure):
 subprocess installed post-installation script returned error exit status 1
Errors were encountered while processing:
 initramfs-tools
E: Sub-process /usr/bin/dpkg returned an error code (1)

I have tried the rpi-source solution mentioned above but it didn't help. Any suggestions?