abperiasamy/rtl8812AU_8821AU_linux #76: rdesktop is not connecting or freezes after login

ionutr2015, 2015-07-20:
Hi,

I'm using your driver on kernel 4.0.x  and I have trouble with rdesktop when using 5GHz (802.11ac). rdesktop is not connecting at all or if it's connecting it freezes most of the times 
after login.
Any idea on how to fix it ?

thanks

joseph-jja, 2015-07-21:
Device? How about lsusb output? iw list output too?

ionutr2015, 2015-07-21:
Device is Edimax EW-7811UTC.

% lsusb
Bus 004 Device 005: ID 0458:004a KYE Systems Corp. (Mouse Systems) 
Bus 004 Device 004: ID 046d:c313 Logitech, Inc. Internet 350 Keyboard
Bus 004 Device 003: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 004 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 004: ID 058f:6366 Alcor Micro Corp. Multi Flash Reader
Bus 003 Device 003: ID 7392:a812 Edimax Technology Co., Ltd 
Bus 003 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

% iw list
iw list
Wiphy phy0
	max # scan SSIDs: 9
	max scan IEs length: 2304 bytes
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Supported Ciphers:
		* WEP40 (00-0f-ac:1)
		* WEP104 (00-0f-ac:5)
		* TKIP (00-0f-ac:2)
		* CCMP (00-0f-ac:4)
	Available Antennas: TX 0 RX 0
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * monitor
		 * P2P-client
		 * P2P-GO
	Band 1:
		Capabilities: 0x1862
			HT20/HT40
			Static SM Power Save
			RX HT20 SGI
			RX HT40 SGI
			No RX STBC
			Max AMSDU length: 7935 bytes
			DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 16 usec (0x07)
		HT TX/RX MCS rate indexes supported: 0-7, 32
		Bitrates (non-HT):
			* 1.0 Mbps
			* 2.0 Mbps
			* 5.5 Mbps
			* 11.0 Mbps
			* 6.0 Mbps
			* 9.0 Mbps
			* 12.0 Mbps
			* 18.0 Mbps
			* 24.0 Mbps
			* 36.0 Mbps
			* 48.0 Mbps
			* 54.0 Mbps
		Frequencies:
			* 2412 MHz [1] (20.0 dBm)
			* 2417 MHz [2] (20.0 dBm)
			* 2422 MHz [3] (20.0 dBm)
			* 2427 MHz [4] (20.0 dBm)
			* 2432 MHz [5] (20.0 dBm)
			* 2437 MHz [6] (20.0 dBm)
			* 2442 MHz [7] (20.0 dBm)
			* 2447 MHz [8] (20.0 dBm)
[... 104 lines cut ...]
		 * del_pmksa
		 * flush_pmksa
		 * remain_on_channel
		 * frame
		 * set_channel
		 * connect
		 * disconnect
	Supported TX frame types:
		 * IBSS: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * managed: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * AP: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * AP/VLAN: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-client: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-GO: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
	Supported RX frame types:
		 * IBSS: 0xd0
		 * managed: 0x40 0xd0
		 * AP: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * AP/VLAN: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * P2P-client: 0x40 0xd0
		 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
	software interface modes (can always be added):
		 * monitor
	interface combinations are not supported
	Device supports scan flush.

joseph-jja, 2015-07-21:
So this driver is the Edimax driver with some fixes applied so it compiles on newer kernels. 

Does rdesktop work on 2.4ghz?

ionutr2015, 2015-07-21:
It's exactly the driver from this git. No need for fixes to compile on newer kernels.\

Yes, it's working on 2.4 GHz, although it's a little slow.

joseph-jja, 2015-07-21:
This git repo is the Edimax driver, from their website.  It was added to git and has had some patches so it compiles on newer kernels and some other compile fixes.

ionutr2015, 2015-07-21:
Yes, I know. I wanted to say that I didn't apply any other patches.

ionutr2015, 2015-07-21:
If I use 40 MHz channel for 2.4 GHz (150 mbps), I get freezes on 2.4 GHz after about 5 minutes of working.

joseph-jja, 2015-07-21:
What channel freq do you use on the 5ghz?

ionutr2015, 2015-07-21:
I'm using channel 36.

Btw,  everthing was ok on kernel 3.x.x from Ubuntu LTS 14.04. I was using the driver from gnab git  (https://github.com/gnab/rtl8812au.git) then I switched to fedora 22 and used this driver. Trouble started with kernel 4.0.x.

harshavardhana, 2015-07-21:
> (https://github.com/gnab/rtl8812au.git) then I switched to fedora 22 and used this driver. Trouble started with kernel 4.0.x.

4.0.x had lot of low level changes in wireless subsystem - it was hard time getting things compiled properly, which could be possibly the reason that there are some changes which are missing.  

Most of our tests are largely on 3.x.x for Ubuntu 14.04, 15.04 .. LTS

ionutr2015, 2015-07-21:
I've actuaĺly been able to compile gnab driver on 4.0.x too with a small change,  but the behavior was exactly the same.

joseph-jja, 2015-07-22:
I had problems on my raspberry pi running 4.0.7+ and this driver.  I found out the problems seem to be exacerbated by a line I added to the file /lib/udev/rules.d/75-persistent-net-generator.rules and kernel 4.x

In that file on the pi it looks something like this

KERNEL!="eth*|ath*|msh*|ra*|sta*|ctc*|lcs*|hsi*",    GOTO="persistent_net_generator_end"

When I changed it to this

KERNEL!="eth*|ath*|wlan*[0-9]|msh*|ra*|sta*|ctc*|lcs*|hsi*",    GOTO="persistent_net_generator_end"

I started having issues.  Ubuntu and I think Fedora do this by default, but Ubuntu 14. is on 3.x kernel.

Try removing the wlan*[0-9]|

ionutr2015, 2015-07-22:
Are those interface names ? In my case the interface name is very different:
wlp0s26u1u2  IEEE 802.11AC  ESSID:"XXX"  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency:5.18 GHz  Access Point: XX:XX:XX:XX:XX:XX
          Bit Rate:434 Mb/s   Sensitivity:0/0

ionutr2015, 2015-07-22:
I like that I can work with rdesktop on 5GHz for the whole day. That's of course if I succeed to connect and pass the login step. The rate for rdesktop to connect is about 10-15% and the rate to pass the login step is also 10-15%

joseph-jja, 2015-07-22:
Yes, that would be interface names.  I wonder if fedora is using systemd?

ionutr2015, 2015-07-23:
Yes, fedora is using systemd.

joseph-jja, 2015-07-25:
Have you tried this driver?

https://github.com/ulli-kroll/rtl8821au

ionutr2015, 2015-07-28:
Thanks, I've tried it. It looks like it's about the same.

ionutr2015, 2015-07-28:
No, actually it's worst. I get freezes all the time.

joseph-jja, 2015-08-12:
It seems there are repos poping up all over github for this driver.  I forked this repo to try a few things.  The latest changes here https://github.com/joseph-jja/rtl8812AU_8821AU_linux are taken from another repo that seems to have a fix for some delays in the driver.  Those delays may be the cause of your issue.  Try my branch and if that works for you maybe we can push the changes up to this branch.

ionutr2015, 2015-08-12:
Thanks, it is still have trouble connecting. 1 of 3 connections are successful but there are no longer login screen freezes

joseph-jja, 2015-08-12:
So your connection success rate is up from 15%-20% to 33%?  No more freezes is good.

ionutr2015, 2015-08-12:
On kernel-4.1.3-201 rate is 25%, on kernel-4.1.3-200 rate is 33%. Overall the things are much better.

harshavardhana, 2015-08-13:
can this be tested again after @joseph-jja fix on #81 ?

ionutr2015, 2015-08-13:
Previous version seems better, although I got freezes on login screen again and I had to try for 30 times to connect until I have succeeded.