abperiasamy/rtl8812AU_8821AU_linux #82: Archer T4UH USB ID

TerribleTadpole, 2015-08-22:
I have successfully used this driver package with the Archer T4UH USB device:
https://wikidevi.com/wiki/TP-LINK_Archer_T4UH

Device ID is 2357:0103

Recommend adding ID to usb_intf.c:
{USB_DEVICE(0x2357, 0x0103),.driver_info = RTL8812}, /* TP-Link - Archer T4UH */

harshavardhana, 2015-08-22:
> {USB_DEVICE(0x2357, 0x0103),.driver_info = RTL8812}, /* TP-Link - Archer T4UH */

would you mind sending a pull request?

TerribleTadpole, 2015-08-23:
Happy to, but...
I didn't create a fork originally and now that I have forked it I note that the device has already been added to the head branch.

harshavardhana, 2015-08-25:
> I didn't create a fork originally and now that I have forked it I note that the device has already been added to the head branch.

cool

closed 2015-08-25 by harshavardhana