abperiasamy/rtl8812AU_8821AU_linux #84: Modprobe, Exec format error

einsiol, 2015-08-25:
I'm not sure were the issue lies, but wanted to reported here as as well.

After my latest system-update on my UbuntuGnome 15.04 (dist-update) I tried to rebuild the driver from this repository like I have been doing for a while now with out any issue. My process being so:
```
make clean
make
sudo make install
sudo modprobe 8812au
```

The issue comes in the end, were I get a error when trying to enable the the driver with modprobe.
```
 modprobe: ERROR: could not insert '8812au': Exec format error 
```

I’ve posted a thread on Ubuntu forums as well http://ubuntuforums.org/showthread.php?t=2292112

harshavardhana, 2015-08-25:
> modprobe: ERROR: could not insert '8812au': Exec format error

Doesn't seem to happen to me

```
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /home/harsha/repos/rtl8812AU_8821AU_linux/8812au.mod.o
  LD [M]  /home/harsha/repos/rtl8812AU_8821AU_linux/8812au.ko
make[1]: Leaving directory '/usr/src/linux-headers-3.19.0-26-generic'
11:54:05: harsha@sysrq:~/repos/rtl8812AU_8821AU_linux(master)$ make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/3.19.0-26-generic/build M=/home/harsha/repos/rtl8812AU_8821AU_linux  modules
make[1]: Entering directory '/usr/src/linux-headers-3.19.0-26-generic'
  Building modules, stage 2.
  MODPOST 1 modules
make[1]: Leaving directory '/usr/src/linux-headers-3.19.0-26-generic'
11:54:12: harsha@sysrq:~/repos/rtl8812AU_8821AU_linux(master)$ sudo make install
install -p -m 644 8812au.ko  /lib/modules/3.19.0-26-generic/kernel/drivers/net/wireless/
/sbin/depmod -a 3.19.0-26-generic
11:55:31: harsha@sysrq:~/repos/rtl8812AU_8821AU_linux(master)$ sudo modprobe 8812au 
11:58:55: harsha@sysrq:~/repos/rtl8812AU_8821AU_linux(master)$ 
```

Can you try removing 

```
rm /lib/modules/$(uname -r)/kernel/drivers/net/wireless/8812au.ko
```

and recompiling it again?

einsiol, 2015-08-25:
Just tried that and getting same result. I also want to note that I'm using 3.19.0-27-generic

harshavardhana, 2015-08-25:
> Just tried that and getting same result. I also want to note that I'm using 3.19.0-27-generic

Yep tried that too, can install and use it properly. 

```
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /home/harsha/repos/rtl8812AU_8821AU_linux/8812au.mod.o
  LD [M]  /home/harsha/repos/rtl8812AU_8821AU_linux/8812au.ko
make[1]: Leaving directory '/usr/src/linux-headers-3.19.0-27-generic'
13:49:48: harsha@sysrq:~/repos/rtl8812AU_8821AU_linux(master)$ sudo make install
install -p -m 644 8812au.ko  /lib/modules/3.19.0-27-generic/kernel/drivers/net/wireless/
/sbin/depmod -a 3.19.0-27-generic
13:49:52: harsha@sysrq:~/repos/rtl8812AU_8821AU_linux(master)$ sudo modprobe 8812au
```

harshavardhana, 2015-08-25:
For debugging can you do 

```
file 8812au.ko
file /lib/modules/$(uname -r)/kernel/drivers/net/wireless/8812au.ko

modprobe -r 8812au
insmod 8812au.ko
dmesg | tail
```

einsiol, 2015-08-26:
All my efforts have failed, I even tried to revert to the old kernel and it still failed. I think this might just be isolated to my self. I have to try to reformat my machine as I need to get this working ASAP. Thank you for your help.

closed 2015-08-26 by einsiol

harshavardhana, 2015-08-26:
> All my efforts have failed, I even tried to revert to the old kernel and it still failed. I think this might just be isolated to my self. I have to try to reformat my machine as I need to get this working ASAP. Thank you for your help.

quite possibly some corruption on your local system, dmesg would have some indication on what is causing the issue. 

Perhaps you can run an fsck on the boot and root volume in rescue mode which might help fix any known problems.

SafetyCaptain, 2015-12-19:
I have found that modules may still work when forced e.g modprobe --force 8812au.ko

edupsousa, 2016-01-14:
Had the same problem once, was caused by the missing Module.symvers file in kernel source.

stubb0rnCoder, 2016-02-24:
I ran into the same problem as einsiol.
I did a dist-upgrade (Ubuntu-Gnome 15.10, Kernel 4.2.0-30-generic) and after reboot I cannot load the driver.
 
If I run 
`dmesg | tail`
I get the following result
`8812au: disagrees about version of symbol module_layout`

unfortunately I do not have the necessary insight into linux driver and kernel to use this information.

harshavardhana, 2016-02-24:
> I did a dist-upgrade (Ubuntu-Gnome 15.10, Kernel 4.2.0-30-generic) and after reboot I cannot load the driver.

Ubuntu might not work at all 15.10 has never been tested with this driver.

stubb0rnCoder, 2016-02-24:
Shame on me >.<
I panicked before even trying to rebuild the source. 
`make
make install
modprobe 8812au`
did the charm.

OffTopic: The driver worked for me so far. Only issue I encountered yet is that I have to manually modpobe the driver after I booted my dual boot Win10.

harshavardhana, 2016-02-24:
> OffTopic: The driver worked for me so far. Only issue I encountered yet is that I have to manually modpobe the driver after I booted my dual boot Win10.

Manual build is the way right now, dkms was supposed to help but don't know why it is not working.

jglynn43, 2016-02-25:
I had the same symptoms as stubb0rnCoder but I've been using Ubuntu-KDE 15.10 for a while now. It was the last kernel update that did it. It was working fine for me for about two months before that with multiple previous kernel updates. I've been using dkms too and this is the first problem I've encountered.

kwrobel, 2016-03-07:
I am on Debian 8.0 and DKMS way worked from a vanilla installation. Then I added jessie-backports to get newer packages installed (mainly for LibreOffice). Did an update and then dist-upgrade, which installed a much newer backports kernel (nice), and I was of course expecting DKMS to take care of my 8812au module, which it did. But after rebooting, I was stuck without Internet because the module does not in-fact start because of the same error messages reported above. I will try uninstalling from DKMS and building by hand. Will report back.

kwrobel, 2016-03-07:
Yup. So, in removed the DKMS-build version that cannot be modprobe'ed. I ran the following (as root)
dkms remove rtl8812au/1.0 --all (deletes the module for all kernels; I renamed mine to rtl8812au which is a symlink in /usr/src pointing to the unpacked version of the zip archive that I put under /usr/local/src)
I then unpacked the source in my user's home directory, did a make and sudo make install, then sudo modprobe 8812au. That worked. Then I ran sudo rmmod 8812au and sudo make uninstall to get get of the hand-compiled module. I then essentially added the module back to DKMS (from the locations that I had set previously). So I did sudo dkms add rtl8812au/1.0, then sudo dkms build rtl8812au/1.0 and sudo dkms install rtl8812au/1.0. All without errors. Then I ran sudo mobprobe 8812au and the module gets loaded properly this time.

The only thing I can think of what went wrong in the upgrade is, since it was a dist-upgrade and a lot of packages got upgraded to newer versions, maybe the upgrade process compiled the module for the new kernel version while there were still older C libraries in play. That would explain the "not agreeing on symbols" error message. I could test this again. Maybe all that needs to be done in to simply run the build and install portions of dkms again. I'd have to get the machine back to a state without all of the upgraded packages to retest, but the machine I did this on is a very old laptop and it is sloooooooow.

So hope my little blurb helps someone.

sbalmer, 2016-05-09:
After a `dist-upgrade` the module did not load anymore, but it would load with `modprobe --force 8812au`. I then did all of `dkms remove add build install` and the module now works again. 

Thanks everybody for the helpful comments that allowed me to resolve this quickly.

brunocassol, 2017-09-12:
@kwrobel 's suggestion fixed it for me.

I have `Lubuntu 16.04` with kernel `4.4.0-93-generic` and USB wifi adapter `Edimax EW-7811UAC AC600`. I'm using this driver: https://github.com/diederikdehaas/rtl8812AU. Works like a charm. Very stable.

edzob, 2017-11-05:
For me worked the following to not get the error message.. 
The (lack of the) prefix RTL was the thing that gave the error on my machine

rm /lib/modules/$(uname -r)/kernel/drivers/net/wireless/rtl8812au.ko
make clean
make
sudo make install
sudo modprobe rtl8812au

tanweer-avg, 2017-11-23:
@edzob Thanks mate

following your commands worked me like a charm

brunocassol, 2017-11-24:
Also try

`make CC=/usr/bin/gcc-5`

Instead of just make

Jarrux, 2018-01-19:
I have the same problem due to a kernel update from 4.10.0-28-generic to 4.13.-26-generic

Unable to compile under the new kernel. I came back to older version, and everything was fine.

doa379, 2018-06-02:
Built and working well with Kernel 4.14.37.

farran, 2019-10-23:
@kwrobel your description was really useful to me, thank you. I was struggling with a different wifi chip driver, but through your guide, I managed to get it installed so that it didn't keep deactivating, and with DKMS so that I have now re-enabled secure boot and the driver still loads. Thank you so much :)

ant19881, 2019-11-16:
git clone https://github.com/scrivy/rtl8812AU_8821AU_linux.git
sudo modprobe -a rtl8812au
***********
OPTIONS
       -a, --all
           Insert all module names on the command line