abperiasamy/rtl8812AU_8821AU_linux #86: Help with bad performance

wis66, 2015-09-05:
Hello,

First of all, thanks for providing this driver and for maintaining it.

I have recently installed Linux on my machine.
I was able to activate the wifi thanks to this driver, but the performance is quite bad compared to the one obtained with Windows. The download speed is around 1-4 Mbps where I am usually around 50Mbps with Windows. Surprisingly the upload speed is better at 5-10 Mbps.

The card is embedded in the computer. Its ID is 0bda:a811 as given by lsusb (this ID is mentionned in issue #71 and pull request #72 as a cheap device but the computer it is embedded in is not ... it is an ASUS ROG GR8).

Do you have any ideas on how to improve performance?
I'd be happy to run some more diagnostics ...

Thanks.

PS: I have tried several other versions available on github with not much luck.
I have also tried to contact ASUS and realtek about this issue but am not very confident in their responses.

harshavardhana, 2015-09-05:
> PS: I have tried several other versions available on github with not much luck.
> I have also tried to contact ASUS and realtek about this issue but am not very confident in their responses.

Can you attach your lspci and lsusb information here?

wis66, 2015-09-06:
Sure.

lsusb
Bus 003 Device 002: ID 8087:8000 Intel Corp. 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 0bda:a811 Realtek Semiconductor Corp. 
Bus 001 Device 003: ID 046d:c05a Logitech, Inc. M90/M100 Optical Mouse
Bus 001 Device 002: ID 045e:07f8 Microsoft Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

lspci
00:00.0 Host bridge: Intel Corporation Haswell-ULT DRAM Controller (rev 0b)
00:02.0 VGA compatible controller: Intel Corporation Haswell-ULT Integrated Graphics Controller (rev 0b)
00:03.0 Audio device: Intel Corporation Haswell-ULT HD Audio Controller (rev 0b)
00:14.0 USB controller: Intel Corporation Lynx Point-LP USB xHCI HC (rev 04)
00:16.0 Communication controller: Intel Corporation Lynx Point-LP HECI #0 (rev 04)
00:19.0 Ethernet controller: Intel Corporation Ethernet Connection I218-V (rev 04)
00:1b.0 Audio device: Intel Corporation Lynx Point-LP HD Audio Controller (rev 04)
00:1c.0 PCI bridge: Intel Corporation Lynx Point-LP PCI Express Root Port 1 (rev e4)
00:1c.4 PCI bridge: Intel Corporation Lynx Point-LP PCI Express Root Port 5 (rev e4)
00:1d.0 USB controller: Intel Corporation Lynx Point-LP USB EHCI #1 (rev 04)
00:1f.0 ISA bridge: Intel Corporation Lynx Point-LP LPC Controller (rev 04)
00:1f.2 SATA controller: Intel Corporation Lynx Point-LP SATA Controller 1 [AHCI mode] (rev 04)
00:1f.3 SMBus: Intel Corporation Lynx Point-LP SMBus Controller (rev 04)
02:00.0 3D controller: NVIDIA Corporation GM107M [GeForce GTX 860M] (rev a2)