abperiasamy/rtl8812AU_8821AU_linux #87: fix-some-compile-warning-for-kernel-4.1

ptpt52, 2015-09-08:
hello, if you accept I world like to commit to your repository.

From 6aaef99f2d3dc181c83d0ad3019a1d85d172dbcc Mon Sep 17 00:00:00 2001
From: Chen Minqiang <ptpt52@gmail.com>
Date: Tue, 8 Sep 2015 22:26:33 +0800
Subject: [PATCH] fix some compile warning for kernel 4.1

---
 core/rtw_sta_mgt.c            |  6 +++---
 include/osdep_service.h       |  2 +-
 include/sta_info.h            |  4 ++--
 include/wifi.h                |  2 +-
 os_dep/linux/ioctl_cfg80211.c | 28 +++++++++++++++++++++++++---
 os_dep/linux/rtw_android.c    | 14 +++++++++++++-
 os_dep/osdep_service.c        |  2 +-
 7 files changed, 46 insertions(+), 12 deletions(-)

diff --git a/core/rtw_sta_mgt.c b/core/rtw_sta_mgt.c
index 889ca76..56ff1fe 100644
--- a/core/rtw_sta_mgt.c
+++ b/core/rtw_sta_mgt.c
@@ -690,7 +690,7 @@ _func_exit_;
 }
 
 /* any station allocated can be searched by hash list */
-struct sta_info *rtw_get_stainfo(struct sta_priv *pstapriv, u8 *hwaddr)
+struct sta_info *rtw_get_stainfo(struct sta_priv *pstapriv, const u8 *hwaddr)
 {
 
 	_irqL	 irqL;
@@ -701,9 +701,9 @@ struct sta_info *rtw_get_stainfo(struct sta_priv *pstapriv, u8 *hwaddr)
 	
 	u32	index;
 
-	u8 *addr;
+	const u8 *addr;
 
-	u8 bc_addr[ETH_ALEN] = {0xff,0xff,0xff,0xff,0xff,0xff};
+	const u8 bc_addr[ETH_ALEN] = {0xff,0xff,0xff,0xff,0xff,0xff};
 
 _func_enter_;
 
diff --git a/include/osdep_service.h b/include/osdep_service.h
index 050967a..82fa89c 100644
--- a/include/osdep_service.h
+++ b/include/osdep_service.h
@@ -206,7 +206,7 @@ extern void	rtw_sleep_schedulable(int ms);
 extern void	rtw_msleep_os(int ms);
 extern void	rtw_usleep_os(int us);
 
-extern u32 	rtw_atoi(u8* s);
+extern u32 	rtw_atoi(const u8* s);
 
 #ifdef DBG_DELAY_OS
 #define rtw_mdelay_os(ms) _rtw_mdelay_os((ms), __FUNCTION__, __LINE__)
diff --git a/include/sta_info.h b/include/sta_info.h
index beaa3be..fe0b92b 100644
--- a/include/sta_info.h
+++ b/include/sta_info.h
@@ -425,7 +425,7 @@ struct	sta_priv {
 };
 
 
-__inline static u32 wifi_mac_hash(u8 *mac)
+__inline static u32 wifi_mac_hash(const u8 *mac)
 {
         u32 x;
 
@@ -453,7 +453,7 @@ struct sta_info *rtw_get_stainfo_by_offset(struct sta_priv *stapriv, int offset)
 extern struct sta_info *rtw_alloc_stainfo(struct	sta_priv *pstapriv, u8 *hwaddr);
 extern u32	rtw_free_stainfo(_adapter *padapter , struct sta_info *psta);
 extern void rtw_free_all_stainfo(_adapter *padapter);
-extern struct sta_info *rtw_get_stainfo(struct sta_priv *pstapriv, u8 *hwaddr);
+extern struct sta_info *rtw_get_stainfo(struct sta_priv *pstapriv, const u8 *hwaddr);
 extern u32 rtw_init_bcmc_stainfo(_adapter* padapter);
 extern struct sta_info* rtw_get_bcmc_stainfo(_adapter* padapter);
 extern u8 rtw_access_ctrl(_adapter *padapter, u8 *mac_addr);
diff --git a/include/wifi.h b/include/wifi.h
index 162e0d9..b42bcbc 100644
--- a/include/wifi.h
+++ b/include/wifi.h
@@ -428,7 +428,7 @@ enum WIFI_REG_DOMAIN {
 		(addr[4] == 0xff) && (addr[5] == 0xff) )  ? _TRUE : _FALSE \
 )
 
-__inline static int IS_MCAST(unsigned char *da)
+__inline static int IS_MCAST(const unsigned char *da)
 {
 	if ((*da) & 0x01)
 		return _TRUE;
diff --git a/os_dep/linux/ioctl_cfg80211.c b/os_dep/linux/ioctl_cfg80211.c
index d1ba336..4946de3 100644
--- a/os_dep/linux/ioctl_cfg80211.c
+++ b/os_dep/linux/ioctl_cfg80211.c
@@ -1382,7 +1382,7 @@ static int cfg80211_rtw_set_default_key(struct wiphy *wiphy,
 
 static int cfg80211_rtw_get_station(struct wiphy *wiphy,
 				    struct net_device *ndev,
-				    u8 *mac, struct station_info *sinfo)
+				    const u8 *mac, struct station_info *sinfo)
 {
 	int ret = 0;
 	_adapter *padapter = wiphy_to_adapter(wiphy);
@@ -2488,7 +2488,7 @@ static int cfg80211_rtw_connect(struct wiphy *wiphy, struct net_device *ndev,
 	NDIS_802_11_AUTHENTICATION_MODE authmode;
 	NDIS_802_11_SSID ndis_ssid;
 	u8 *dst_ssid, *src_ssid;
-	u8 *dst_bssid, *src_bssid;
+	const u8 *dst_bssid, *src_bssid;
 	//u8 matched_by_bssid=_FALSE;
 	//u8 matched_by_ssid=_FALSE;
 	u8 matched=_FALSE;
@@ -3331,7 +3331,11 @@ static const struct net_device_ops rtw_cfg80211_monitor_if_ops = {
 };
 #endif
 
+#if (LINUX_VERSION_CODE >= KERNEL_VERSION(4,1,0))
+static int rtw_cfg80211_add_monitor_if(_adapter *padapter, char *name, unsigned char name_assign_type, struct net_device **ndev)
+#else
 static int rtw_cfg80211_add_monitor_if(_adapter *padapter, char *name, struct net_device **ndev)
+#endif
 {
 	int ret = 0;
 	struct net_device* mon_ndev = NULL;
@@ -3362,6 +3366,9 @@ static int rtw_cfg80211_add_monitor_if(_adapter *padapter, char *name, struct ne
 	mon_ndev->type = ARPHRD_IEEE80211_RADIOTAP;
 	strncpy(mon_ndev->name, name, IFNAMSIZ);
 	mon_ndev->name[IFNAMSIZ - 1] = 0;
+#if (LINUX_VERSION_CODE >= KERNEL_VERSION(4,1,0))
+	mon_ndev->name_assign_type = name_assign_type;
+#endif
 	mon_ndev->destructor = rtw_ndev_destructor;
 
 #if (LINUX_VERSION_CODE>=KERNEL_VERSION(2,6,29))
@@ -3426,6 +3433,9 @@ static int
 	#else
 		char *name,
 	#endif
+	#if (LINUX_VERSION_CODE >= KERNEL_VERSION(4,1,0))
+		unsigned char name_assign_type,
+	#endif
 		enum nl80211_iftype type, u32 *flags, struct vif_params *params)
 {
 	int ret = 0;
@@ -3443,7 +3453,11 @@ static int
 		ret = -ENODEV;
 		break;
 	case NL80211_IFTYPE_MONITOR:
+	#if (LINUX_VERSION_CODE >= KERNEL_VERSION(4,1,0))
+		ret = rtw_cfg80211_add_monitor_if(padapter, (char *)name, name_assign_type, &ndev);
+	#else
 		ret = rtw_cfg80211_add_monitor_if(padapter, (char *)name, &ndev);
+	#endif
 		break;
 
 #if (LINUX_VERSION_CODE >= KERNEL_VERSION(2,6,37)) || defined(COMPAT_KERNEL_RELEASE)
@@ -3713,13 +3727,21 @@ static int	cfg80211_rtw_add_station(struct wiphy *wiphy, struct net_device *ndev
 	return 0;
 }
 
+#if (LINUX_VERSION_CODE >= KERNEL_VERSION(4,1,0))
+static int cfg80211_rtw_del_station(struct wiphy *wiphy, struct net_device *ndev, struct station_del_parameters *params)
+#else
 static int	cfg80211_rtw_del_station(struct wiphy *wiphy, struct net_device *ndev,
 #if (LINUX_VERSION_CODE >= KERNEL_VERSION(3,11,0)) && (LINUX_VERSION_CODE < KERNEL_VERSION(3,16,0)) && !defined(COMPAT_KERNEL_RELEASE)
 			       u8 *mac)
 #else
 			       const u8 *mac)
 #endif
+#endif
 {
+
+#if (LINUX_VERSION_CODE >= KERNEL_VERSION(4,1,0))
+	const u8 *mac = params->mac;
+#endif
 	int ret=0;
 	_irqL irqL;
 	_list	*phead, *plist;
@@ -3809,7 +3831,7 @@ static int	cfg80211_rtw_del_station(struct wiphy *wiphy, struct net_device *ndev
 }
 
 static int	cfg80211_rtw_change_station(struct wiphy *wiphy, struct net_device *ndev,
-#if (LINUX_VERSION_CODE >= KERNEL_VERSION(3,11,0)) && !defined(COMPAT_KERNEL_RELEASE)
+#if (LINUX_VERSION_CODE >= KERNEL_VERSION(3,11,0)) && !defined(COMPAT_KERNEL_RELEASE) && (LINUX_VERSION_CODE <= KERNEL_VERSION(4,0,0))
 				  u8 *mac, struct station_parameters *params)
 #else
 				  const u8 *mac, struct station_parameters *params)
diff --git a/os_dep/linux/rtw_android.c b/os_dep/linux/rtw_android.c
index 6f7938e..e972a7f 100644
--- a/os_dep/linux/rtw_android.c
+++ b/os_dep/linux/rtw_android.c
@@ -308,7 +308,7 @@ int rtw_android_set_block(struct net_device *net, char *command, int total_len)
 	return 0;
 }
 
-int get_int_from_command(char* pcmd )
+int get_int_from_command(const char* pcmd )
 {
 	int i = 0;
 
@@ -554,7 +554,11 @@ int rtw_android_priv_cmd(struct net_device *net, struct ifreq *ifr, int cmd)
 	
 		pwfd_info = &padapter->wfd_info;
 		if( padapter->wdinfo.driver_interface == DRIVER_CFG80211 )
+#ifdef CONFIG_COMPAT
+			pwfd_info->rtsp_ctrlport = ( u16 ) get_int_from_command( compat_ptr(priv_cmd.buf) );
+#else
 			pwfd_info->rtsp_ctrlport = ( u16 ) get_int_from_command( priv_cmd.buf );
+#endif
 		break;
 	}
 	case ANDROID_WIFI_CMD_WFD_SET_MAX_TPUT:
@@ -574,7 +578,11 @@ int rtw_android_priv_cmd(struct net_device *net, struct ifreq *ifr, int cmd)
 		pwfd_info = &padapter->wfd_info;
 		if( padapter->wdinfo.driver_interface == DRIVER_CFG80211 )
 		{
+#ifdef CONFIG_COMPAT
+			pwfd_info->wfd_device_type = ( u8 ) get_int_from_command( compat_ptr(priv_cmd.buf) );
+#else
 			pwfd_info->wfd_device_type = ( u8 ) get_int_from_command( priv_cmd.buf );
+#endif
 		
 			pwfd_info->wfd_device_type &= WFD_DEVINFO_DUAL;
 		}
@@ -598,7 +606,11 @@ response:
 			bytes_written++;
 		}
 		priv_cmd.used_len = bytes_written;
+#ifdef CONFIG_COMPAT
+		if (copy_to_user(compat_ptr(priv_cmd.buf), command, bytes_written)) {
+#else
 		if (copy_to_user((void *)priv_cmd.buf, command, bytes_written)) {
+#endif
 			DBG_871X("%s: failed to copy data to user buffer\n", __FUNCTION__);
 			ret = -EFAULT;
 		}
diff --git a/os_dep/osdep_service.c b/os_dep/osdep_service.c
index fc68ad1..3aef6a9 100644
--- a/os_dep/osdep_service.c
+++ b/os_dep/osdep_service.c
@@ -55,7 +55,7 @@ inline int RTW_STATUS_CODE(int error_code){
 }
 #endif
 
-u32 rtw_atoi(u8* s)
+u32 rtw_atoi(const u8* s)
 {
 
 	int num=0,flag=0;
-- 
1.9.1

harshavardhana, 2015-09-08:
@ptpt52 - thanks for the patch, please send a pull request - http://yangsu.github.io/pull-request-tutorial/

harshavardhana, 2015-09-09:
merged in #88

closed 2015-09-09 by harshavardhana