abperiasamy/rtl8812AU_8821AU_linux #89: ap mode with wpa not working

ptpt52, 2015-09-09:
tools: create_ap shell script for fast create ap mode use the adapter
git clone https://github.com/oblique/create_ap.git

what is the issue:

   sudo insmod ./8812au.ko
   sudo ./create_ap -n --freq-band 5 -c 36 -w 2 wlan1 TEST_AP_WIFI mypassword
first time the ap "TEST_AP_WIFI" work good, my phone can connect to the 'TEST_AP_WIFI'
and hit Ctrl + C to stop the ./create_ap....
and do it again
   sudo ./create_ap -n --freq-band 5 -c 36 -w 2 wlan1 TEST_AP_WIFI mypassword
for now the ap shows the SSID 'TEST_AP_WIFI' but my phone can never connect to it.

that is the issue.

harshavardhana, 2015-09-09:
> sudo ./create_ap -n --freq-band 5 -c 36 -w 2 wlan1 TEST_AP_WIFI mypassword
> first time the ap "TEST_AP_WIFI" work good, my phone can connect to the 'TEST_AP_WIFI'
> and hit Ctrl + C to stop the [...]

The issue here is perhaps almost like the first attempt gets the driver registered properly but second attempt is really not cleaning the previous state. 

Personally i have never tried this, can you paste some debugging output from 'dmesg' or syslog?

piskorzj, 2018-05-17:
@ptpt52 Which device you used to set up AP on a 5Ghz band? From my research, it seems that it has to support setting regulatory domain (instead of having hardcoded 00).

kimocoder, 2018-05-17:
And here's how.. look at our commit 14, 15 and 16 over [here](https://github.com/aircrack-ng/rtl8812au/commits/v5.1.5) nuked it and AP working