abperiasamy/rtl8812AU_8821AU_linux #91: Very slow DNS lookup for initial connection

caprica, 2015-09-22:
I have two computers on the same network, one wired to a regular broadband router, the other using an ASUS AC-56 USB device with this driver connected to the same router.

I followed the dkms instructions and everything is *almost* working.

I get something like an 80% signal strength.

The wired PC obviously works just fine, I can browse to web-sites and it's very quick.

The WIFI PC has some strange issues resolving hosts - e.g. if I go direct to www.bbc.co.uk it appears almost instantaneously. If I go to www.google.co.uk, it takes an age and sometimes times out with a DNS lookup failure, whereas other times it will eventually come back and after that point I can use google as normal... for a while... and then it starts timing out again.

If I switch out the AC-56 for a cheap Startech USB dongle in the same PC, that also works just fine and google resolves instantly.

Do you have any tips on what this could be?

crholm, 2015-09-29:
I have an AC56 as well and experience the same problem. To me it seems running 802.11AC is the problem, once i switched to 802.11N everything works fine.

harshavardhana, 2015-09-29:
> I have an AC56 as well and experience the same problem. To me it seems running 802.11AC is the problem, once i switched to 802.11N everything works fine.

EDIMAX 1200-AC works great with 802.11AC with this driver - the problem we have seen is with other wifi cards with AC support.

closed 2020-04-11 by caprica