abperiasamy/rtl8812AU_8821AU_linux #93: D-Link DWA182 Rev C1, tested?

boekhold, 2015-09-28:
Hi all,

I'm looking to upgrade my media PC (running Linux Mint 17.2 with kernel 3.19) with a Wireless AC solution. Basically the **only** Wireless AC USB adapter that is available here (UAE) is the D-Link DWA182 Rev C1.

I realize this isn't really an issue, but I can't find a general discussion forum on this topic. Has anybody used this driver with the D-Link DWA182 Rev C1 and kernel 3.19? Does the dkms stuff work? How is the performance? I've seen a couple of negative reports on actual wireless performance with this chipset.

boekhold, 2015-09-28:
The D-Link website has a version 4.3.2 of this driver. Is that newer/better than this driver? Which one is preferred?

harshavardhana, 2015-09-28:
> The D-Link website has a version 4.3.2 of this driver. Is that newer/better than this driver? Which one is preferred?

I would say D-Link driver would be preferred, we haven't rebased our changes new changes in RTL8812AU  and never tested D-Link DWA182.  What we do have is EDIMAX AC-1200 which works great.