abperiasamy/rtl8812AU_8821AU_linux #96: Crash

pouleta, 2015-10-05:
Hello,
a crash, when using the driver at my school (only there !)

```plain
[  +0,000078] ------------[ cut here ]------------
[  +0,000018] WARNING: CPU: 1 PID: 6 at net/wireless/sme.c:714 __cfg80211_connect_result+0x399/0x410 [cfg80211]()
[  +0,000002] Modules linked in: fuse ax25 snd_hda_codec_hdmi joydev mousedev uvcvideo videobuf2_vmalloc videobuf2_memops videobuf2_core v4l2_common btusb btrtl videodev 8812au(O) btbcm btintel bluetooth media arc4 nvram iTCO_wdt msr iTCO_vendor_support intel_rapl iosf_mbi x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel iwlmvm kvm mac80211 crct10dif_pclmul crc32_pclmul ghash_clmulni_intel aesni_intel i915 aes_x86_64 lrw gf128mul glue_helper ablk_helper cryptd snd_hda_codec_realtek snd_hda_codec_generic evdev iwlwifi input_leds pcspkr mac_hid rtsx_pci_ms drm_kms_helper serio_raw snd_hda_intel memstick psmouse mei_me cfg80211 snd_hda_codec drm r8169 mii i2c_i801 rfkill snd_hda_core mei shpchp lpc_ich intel_gtt snd_hwdep i2c_algo_bit snd_soc_rt5640 fan thermal wmi snd_soc_rl6231 snd_soc_core
[  +0,000064]  battery snd_compress snd_pcm_dmaengine ac97_bus snd_pcm i2c_hid snd_timer dw_dmac hid snd soundcore i2c_designware_platform video dw_dmac_core gpio_lynxpoint snd_soc_sst_acpi tpm_crb 8250_dw i2c_designware_core spi_pxa2xx_platform tpm acpi_pad ac button processor sch_fq_codel ip_tables x_tables ext4 crc16 mbcache jbd2 sr_mod cdrom sd_mod ahci libahci libata crc32c_intel atkbd libps2 xhci_pci ehci_pci rtsx_pci_sdmmc xhci_hcd ehci_hcd scsi_mod usbcore rtsx_pci usb_common i8042 serio sdhci_acpi sdhci led_class mmc_core dm_mod
[  +0,000045] CPU: 1 PID: 6 Comm: kworker/u8:0 Tainted: G        W  O    4.2.1-1-ARCH #1
[  +0,000003] Hardware name: Notebook                         W54_55_94_95_97AU,AUQ/W54_55_94_95_97AU,AUQ, BIOS 5.6.5 01/09/2015
[  +0,000010] Workqueue: cfg80211 cfg80211_event_work [cfg80211]
[  +0,000002]  0000000000000000 00000000e1077fdc ffff880119befc48 ffffffff8156b77a
[  +0,000004]  0000000000000000 0000000000000000 ffff880119befc88 ffffffff81074846
[  +0,000005]  ffff8800d93c1c78 ffff8800d93c1c00 0000000000000000 0000000000000000
[  +0,000004] Call Trace:
[  +0,000006]  [<ffffffff8156b77a>] dump_stack+0x4c/0x6e
[  +0,000005]  [<ffffffff81074846>] warn_slowpath_common+0x86/0xc0
[  +0,000003]  [<ffffffff8107497a>] warn_slowpath_null+0x1a/0x20
[  +0,000008]  [<ffffffffa052fa39>] __cfg80211_connect_result+0x399/0x410 [cfg80211]
[  +0,000007]  [<ffffffff810bb03a>] ? mutex_spin_on_owner.isra.0+0x6a/0x80
[  +0,000008]  [<ffffffffa050746f>] cfg80211_process_wdev_events+0x13f/0x1b0 [cfg80211]
[  +0,000006]  [<ffffffffa0507512>] cfg80211_process_rdev_events+0x32/0x70 [cfg80211]
[  +0,000006]  [<ffffffffa05020f2>] cfg80211_event_work+0x22/0x30 [cfg80211]
[  +0,000005]  [<ffffffff8108c57b>] process_one_work+0x14b/0x440
[  +0,000003]  [<ffffffff8108c8b8>] worker_thread+0x48/0x4a0
[  +0,000002]  [<ffffffff8108c870>] ? process_one_work+0x440/0x440
[  +0,000003]  [<ffffffff81092518>] kthread+0xd8/0xf0
[  +0,000003]  [<ffffffff81092440>] ? kthread_worker_fn+0x170/0x170
[  +0,000003]  [<ffffffff815710df>] ret_from_fork+0x3f/0x70
[  +0,000002]  [<ffffffff81092440>] ? kthread_worker_fn+0x170/0x170
[  +0,000002] ---[ end trace 094bf6381ee7f124 ]---
```

VictorLamoine, 2016-03-03:
Same here, any solution?

```bash
[  378.627103] ------------[ cut here ]------------
[  378.627117] WARNING: CPU: 2 PID: 163 at /build/linux-lts-wily-cCEfLI/linux-lts-wily-4.2.0/net/wireless/sme.c:714 __cfg80211_connect_result+0x39f/0x3e0 [cfg80211]()
[  378.627120] Modules linked in: 8812au(OE) nls_iso8859_1 snd_hda_codec_hdmi input_leds joydev snd_hda_codec_realtek snd_hda_codec_generic snd_hda_intel snd_hda_codec snd_hda_core eeepc_wmi snd_hwdep asus_wmi sparse_keymap mxm_wmi snd_pcm snd_seq_midi intel_rapl snd_seq_midi_event x86_pkg_temp_thermal nvidia(POE) intel_powerclamp coretemp snd_rawmidi snd_seq kvm crct10dif_pclmul snd_seq_device snd_timer crc32_pclmul bnep cfg80211 rfcomm aesni_intel bluetooth aes_x86_64 lrw snd gf128mul glue_helper drm ablk_helper soundcore cryptd mei_me serio_raw shpchp mei lpc_ich 8250_fintek wmi tpm_infineon binfmt_misc video acpi_pad mac_hid parport_pc ppdev lp parport hid_generic usbhid uas hid usb_storage psmouse ahci libahci e1000e ptp pps_core
[  378.627154] CPU: 2 PID: 163 Comm: kworker/u8:4 Tainted: P        W  OE   4.2.0-32-generic #37~14.04.1-Ubuntu
[  378.627155] Hardware name: ASUS All Series/Z97-A, BIOS 2801 11/11/2015
[  378.627160] Workqueue: cfg80211 cfg80211_event_work [cfg80211]
[  378.627164]  ffffffffc0404c80 ffff880036e27c68 ffffffff817b5fed 0000000000000001
[  378.627166]  0000000000000000 ffff880036e27ca8 ffffffff8107799a ffff8801e67a7c78
[  378.627167]  0000000000000000 0000000000000000 ffff8800bbc09818 ffff8801e67a7c00
[  378.627168] Call Trace:
[  378.627172]  [<ffffffff817b5fed>] dump_stack+0x45/0x57
[  378.627175]  [<ffffffff8107799a>] warn_slowpath_common+0x8a/0xc0
[  378.627177]  [<ffffffff81077a8a>] warn_slowpath_null+0x1a/0x20
[  378.627183]  [<ffffffffc03e41df>] __cfg80211_connect_result+0x39f/0x3e0 [cfg80211]
[  378.627188]  [<ffffffffc03befbf>] cfg80211_process_wdev_events+0x13f/0x1b0 [cfg80211]
[  378.627193]  [<ffffffffc03bf062>] cfg80211_process_rdev_events+0x32/0x70 [cfg80211]
[  378.627196]  [<ffffffffc03ba022>] cfg80211_event_work+0x22/0x30 [cfg80211]
[  378.627199]  [<ffffffff8108fb3e>] process_one_work+0x14e/0x3d0
[  378.627201]  [<ffffffff810901ea>] worker_thread+0x11a/0x470
[  378.627206]  [<ffffffff810900d0>] ? rescuer_thread+0x310/0x310
[  378.627208]  [<ffffffff810957a2>] kthread+0xd2/0xf0
[  378.627209]  [<ffffffff810956d0>] ? kthread_create_on_node+0x1c0/0x1c0
[  378.627211]  [<ffffffff817bd81f>] ret_from_fork+0x3f/0x70
[  378.627212]  [<ffffffff810956d0>] ? kthread_create_on_node+0x1c0/0x1c0
[  378.627213] ---[ end trace 9f5198b95e1375c9 ]---
```

VictorLamoine, 2016-03-04:
I have found an other driver that does not expose this bug:
https://github.com/gnab/rtl8812au

closed 2017-12-19 by Nephos