abperiasamy/rtl8812AU_8821AU_linux #97: Monitor Mode not working

vivek29081993, 2015-10-08:
Hi there,
I am using Dual-band Wireless-AC1200 USB 3.0 Wi-Fi Adapter in Kali Linux with kernel version 4.0.0-kali1-amd64.When I use the command :

" airmon-ng start wlan0"

The output is:
-----------------------------------
"No interfering processes found
PHY     Interface       Driver          Chipset
 
 
ethtool failed...
only mac80211 devices on kernel 2.6.33 or higher are officially supported by airmon-ng"

and also the device stops blinking.
The Monitor mode of the adapter is not working using this driver 8812au.
Please help.

fatfarm, 2015-10-21:
ifconfig (interface) down             -------- (interface)example wlan1 
iwconfig (interface) mode monitor
ifconfig (interface) up

proceed to airmon-ng check kill
airodump-ng

okazymyrov, 2015-11-09:
"iwconfig wlan0 mode monitor" doesn't work with D-Link DWA-172. @fatfarm what device do you have?

bits3rpent, 2015-11-15:
This driver does not support monitor mode. It is hardmac-based and therefore most likely will never support monitor mode.

fatfarm, 2015-11-16:
im using the alfa awus036ach on kali linux. the adapter does go into
monitor but wont inject packets.

hackvice, 2019-12-06:
I have the same adapter and I have a hard time with it. Same issues. Goes in monitor mode but does not show any connections at all. On the old driver. When I install the GitHub driver and do upgrade it does not show the wlan0. idk