abperiasamy/rtl8812AU_8821AU_linux #101: Alfa AWUS036ach AC1200

tomchezhl, 2015-10-23:
my distro : LinuxMintDebianEdition = LMDE2 "Betsy" Cinnamon , Kernel 3.16.0-4-amd64
my wlan device : Alfa AWUS036ACH with USB 3.0 (ID 0bda:8812)

1) Several times reported and the same problem on my pc :

bcdUSB 2.0 : if connected to usb 3.0 port the bus assignment is wrong , 
	             if connected to usb 2.0 port the bus assignment is correct.

2) makefile	Config_Platform_I386_PC = y	-> only 32bit driver ???

$ modinfo 8812au
filename:       /lib/modules/3.16.0-4-amd64/kernel/drivers/net/wireless/8812au.ko
version:        v4.2.2_7502.20130517
author:         Realtek Semiconductor Corp.
description:    Realtek Wireless Lan Driver
srcversion:     8844FB5B0B5C170AD1F5D4A

3) Compilation no errors and kernel modul 8812au loadable and wlan device is recognized.
But 8812au is fixed to wlan1 (on my pc).

$ iwconfig
wlan1     unassociated  Nickname:"<WIFI@REALTEK>"

4) AWUS036ach has very bad AP recognition performance. AP needed min. quality 60% to be recogniced. (to compare my AVM Fritz!WLAN (ID 057c:5601) recogniced APs down to a quality of 17% !!!)

5) Do not see any working on 5GHz until yet.

tomchezhl, 2015-10-23:
To Point 1) Several times reported and the same problem on my pc :

bcdUSB 2.0 : if connected to usb 3.0 port the bus assignment is wrong ,
if connected to usb 2.0 port the bus assignment is correct.

I had amail contect to an usb-utils developer https://github.com/gregkh/usbutils .
His conclusion : Unless it's really a USB 2 device there is nothing wrong.

We had to check if it is usb 3.0 or only usb 2.0 which is on the wlan device !!!

trueserve, 2016-05-27:
Regarding weak signal detection, I too am having the same problem with this adapter. Scanning happens extremely quickly. However, unlike tomchezhl, I do see VERY nearby 5GHz networks, but only with the old WLAN interface. Using iw to scan never shows 5GHz results.

All signals reported are extremely weak with iw even if they are nearby. With the old wlan interface they are still reported as very weak but do vary. It's basically unusable.

Device works perfectly fine on Windows.

Gentoo Linux, kernels 3.19.1-gentoo through 4.4.11-gentoo