abperiasamy/rtl8812AU_8821AU_linux #102: 5G packet loss (EW-7822UAC / 8821AU)

mtompkins, 2015-10-30:
Firstly thanks to all involved for their work here. Greatly appreciated.

Current Debian sid (kernel 4.2). 2.4Ghz band works fine.
Solid 300Mb/s connection, latency seems fine.

5G however has crippling packet loss between router and dongle, > 50%.
If there is anything more I can provide to help investigate happy to.

abperiasamy, 2015-10-30:
Hi Mark, does it happen with the official Windows drivers too? 5Ghz is
known for short range limitation.

mtompkins, 2015-10-30:
So I can answer correctly, when you ask about Win drivers, do you mean a Windows machine or dll wrapper for use on Debian?  Also, in respect to range the devices are within arms reach of each other. Other 5GHz devices on the band (Intel AC) are performing well / properly.

mtompkins, 2015-10-30:
This may also provide some helpful info ...

http://pastebin.com/GyCZAxxX

abperiasamy, 2015-10-30:
Yes I meant from a Windows or OS X with an official driver. If your router
is within the same room at line-of-sight, then it is not a signal issue. I
also noticed wlx74da3835e9e9 is connected to 2.462 GHz (Channel 11), which
is 802.11n and not 5GHz "ac" channel. I see 98 packets dropped. Do you
think it is because of channel conflict between three the devices on
channel 11?

      3   APs on   Frequency:2.462 GHz (Channel 11)

mtompkins, 2015-10-30:
Update: I found the following bit of info that is specific to this device - thankfully. Adding it here may prevent someone else going mad.  Testing seems to be quite good once a channel in the range is fixed. Originally I had been on channel 165 and that makes sense to have packet loss of ~ 50%

Edimax wireless adapters support the 5Ghz channels 36, 40, 44, 48, 149, 153, 157, and 161.

mtompkins, 2015-10-30:
Re: You recent post - yes the wlx7... is the Edimax dongle. It is on the 2.4Ghz because of the issue with 5Ghz. Since changing the channel as noted above mtr has it producing wonderful results (no jinx - lol). Will revert / confirm shortly.

mtompkins, 2015-10-30:
OK so I'm comfortable saying things are much improved. Bonded at 867 Mb/s (device max). Zero packet loss. Only thing I notice which is probably outside of this repo driver control is latency between the dongle and the router avg 6ms.  Average on Intel AC 1.3ms (4.2 kernel).

closed 2015-10-30 by mtompkins

abperiasamy, 2015-11-02:
Good to hear. Thanks for posting detailed info. It will be useful for others.