abperiasamy/rtl8812AU_8821AU_linux #104: AP mode dual SSID

Polioman, 2015-11-09:
I'm using an Edimax AC600 USB dongle on a Raspberry and I'd like to know if it's possible to broadcast two SSID using hostapd.

I try to follow [this guide](http://wiki.stocksy.co.uk/wiki/Multiple_SSIDs_with_hostapd) but I can't modify my MAC address. I tried using system tools or macchanger without success.

I've got this error :

```
Network driver didn't actually change to the new MAC!!
```

I really need it and I can not find any clue how to do it.

Reepca, 2016-04-15:
I also can't change the MAC address - not via macchanger (I get the same message as Polioman), ifconfig, or network-manager. Which is a shame, because I'm using this USB adapter since the built-in broadcom stuff can't change it either. Rotten luck, eh. My device is the Edimax AC1200, running on Ubuntu 15.10.

fumantsu, 2016-09-16:
I have the same issue with Dlink DWA-172. There is a module option but doesn't work.