abperiasamy/rtl8812AU_8821AU_linux #105: D-Link DWA-182 - H/W Ver.: C1 - F/W Ver.: 3.0.1

rowaraujo, 2015-11-11:
I have D-Link DWA-182 Rev C1 and and i'm trying to install Kali Linux 2.0 64bits.
I'm using Linux Kali in Virtualbox 5.0.10 with VirtualBox Extension Pack (USB extensions).
I am new in Linux and...

What better way to install my DWA-182?

rowaraujo, 2015-11-11:
I am root and i trying:
# apt-get -y install dkms
# git clone https://github.com/abperiasamy/rtl8812AU_8821AU_linux
# cd rtl8812AU_8821AU_linux
# cp -R . /usr/src/rtl8812AU_8821AU_linux-1.0
# dkms add -m rtl8812AU_8821AU_linux -v 1.0
# dkms build -m rtl8812AU_8821AU_linux -v 1.0
# dkms install -m rtl8812AU_8821AU_linux -v 1.0
# modprobe 8812au

It's correct?

rowaraujo, 2015-11-12:
I installed this driver https://forum.aircrack-ng.org/index.php?topic=1086.0 and it's works now!

okazymyrov, 2015-11-13:
Is there any chance that monitor mode from https://github.com/Grawp/rtl8812au_rtl8821au/tree/4.3.22 will be added to this repository?

rowaraujo, 2015-11-17:
@okazymyrov I don't know... I do not have acess to the repository.