abperiasamy/rtl8812AU_8821AU_linux #106: Doesn't work on ARM Kernel 4.3.0

angelsix, 2015-11-26:
Using upstream kernel 4.3.0 running Armv7l on a Freescale iMX6 board.

Build and install module no problem. Green light comes up on WiFi.

However I get error messages when doing up:

INFO: Possible recursive locking detected
ip/6275 is trying to aquire lock... usbctrl_vendorreq+0x74/0x274
but task is already holding lock: netdev_open+0x24/0x44
..
*** DEADLOCK ***

But it comes up ok.. or it seems to.

However doing:

iwlist wlan0 scan says:
Interface does not support scanning

And iwconfig wlan0 essid XXX key s:XXX says:
Error for wireless request "Set ESSID" (8B1A)
SET Failed on device wlan0: No such file or directory

angelsix, 2015-11-26:
Ok I got it working by using iwconfig instead of ip link to bring it up. However I just noticed, the lsusb -vv shows the info as a "Wireless N" WiFi. This is an AC product is this driver only N support?

joseph-jja, 2015-12-24:
This driver is an AC driver but I have noticed that my adapter always connects to the 2.4 band using N if there is an ssid that has both 2.4 and 5g bands.  I setup a ssid with _5 just for the 5g band and then use that ssid and it connects to the 5g band just fine