abperiasamy/rtl8812AU_8821AU_linux #109: Doesn't work on kernel 4.3

Grief, 2015-12-07:
No errors, but no `wlan0` either. Works well on 4.2.3. Switched several times between kernels - it's always reproducible. Kubuntu 15.10

mtompkins, 2015-12-12:
I've used it on 4.3.0 and currently 4.3.1 - Debian just for info.

morfikov, 2015-12-13:
I can confirm that the module works fine with kernel 4.3 on debian:

    $ uname -a
    Linux morfikownia 4.3.0-trunk-amd64 #1 SMP Debian 4.3-1~exp2 (2015-12-04) x86_64 GNU/Linux

	# dkms install -m rtl8812AU_8821AU_linux -v 1.0

	Kernel preparation unnecessary for this kernel.  Skipping...

	Building module:
	cleaning build area....
	'make'.............................
	cleaning build area....

	DKMS: build completed.

	8812au.ko:
	Running module version sanity check.
	 - Original module
	   - No original module exists within this kernel
	 - Installation
	   - Installing to /lib/modules/4.3.0-trunk-amd64/updates/dkms/

	depmod........

	DKMS: install completed.

	# dkms status
	rtl8812AU_8821AU_linux, 1.0, 4.2.0-1-amd64, x86_64: installed
	rtl8812AU_8821AU_linux, 1.0, 4.3.0-trunk-amd64, x86_64: installed

And the test:

	# /sbin/wpa_supplicant -P /var/run/wpa_supplicant.wlan0.pid -i wlan0 -W -D nl80211,wext -c /etc/wpa_supplicant/wpa_supplicant.conf
	Successfully initialized wpa_supplicant
	WPS: Converting push_button to virtual_push_button for WPS 2.0 compliance
	wlan0: Trying to associate with c4:6e:1f:95:ef:fc (SSID='Valar_Morghulis' freq=5180 MHz)
	wlan0: Associated with c4:6e:1f:95:ef:fc
	wlan0: WPA: Key negotiation completed with c4:6e:1f:95:ef:fc [PTK=CCMP GTK=CCMP]
	wlan0: CTRL-EVENT-CONNECTED - Connection to c4:6e:1f:95:ef:fc completed [id=2 id_str=]
	wlan0: CTRL-EVENT-REGDOM-CHANGE init=COUNTRY_IE type=COUNTRY alpha2=PL

linux-modder, 2015-12-13:
And on 4.4.0-0.rc4.git3.2.fc24.x86_64 (fedora)

----------------------------------------------------------------------------------------------------------------------------------------------

dodgyb, 2015-12-20:
Can confirm dkms build for 4.4, 4.3 and 4.2 is OK on Kubuntu 15.10

linux-modder, 2015-12-20:
4.4.0-0.fc24.x86_64 working  flawless for me here....

@grief,

Care to share the  build log or  other details,   hate to say it but this
seems more like a case of  user unfamiliarity or misunderstanding...

Corey W Sheldon
Freelance IT Consultant, Multi-Discipline Tutor
Ameridea LLC, Founder, CTO
http://github.com/ameridea
Fedora Ambassador, North America
http://getfedora.org
Server Intern Staff, Citygate
http://citygate.org
(p) +1 (310) 909-7672
Find Me  on any of the sites  I teach /frequent:
https://gist.github.com/linux-modder/ac5dc6fa211315c633c9
----------------------------------------------------------------------------------------------------------------------------------------------
    "Have no way as way, no limitation as limitation. One must never
underestimate the power of boredom...from which creativity and laziness are
borne, which can spark great works of chaos and genius."
----------------------------------------------------------------------------------------------------------------------------------------------
PGP: 0xe958c5d6718bf597 FP = 2930 99EB 083D D332 0752 88C4 E958 C5D6 718B
F597 linuxmodder@keybase.io
Tox: Corey84 || Linux-modder
9357BC6A5944A08AFC7D1EFFD61F6A73B9EABF8B2FB84ACF1DAC9A1A4D0A4705FFCCD0E5499B
Linphone: sip:linuxmodder
BitAddress:15cn1BvAFEREHk8UekJ6i9Dxi9Wbw6vzDD

Grief, 2015-12-24:
Sorry, you were right, my fault. Somewhy module was broken: 
```
> sudo insmod /var/lib/dkms/rtl8812AU_8821AU_linux/kernel-4.3.0-040300-generic-x86_64/module/8812au.ko 
insmod: ERROR: could not insert module /var/lib/dkms/rtl8812AU_8821AU_linux/kernel-4.3.0-040300-generic-x86_64/module/8812au.ko: Invalid module format
```
But after I rebuilt it, it works.

closed 2015-12-24 by Grief