abperiasamy/rtl8812AU_8821AU_linux #112: Cannot make on Debian 8 (3.16.0-4-amd64)

Paianni, 2015-12-28:
I tried linking "/lib/modules/3.16.0-4-amd64/build/" to "/usr/src/linux-headers-3.16.0-4-amd64/" but I keep getting the same error:

root@paianni:/home/paianni/Downloads/Packages/rtl8812AU_8821AU_linux# make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/3.16.0-4-amd64/build M=/home/paianni/Downloads/Packages/rtl8812AU_8821AU_linux  modules
make[1]: Entering directory '/lib/modules/3.16.0-4-amd64/build'
make[1]: *** No rule to make target 'modules'.  Stop.
make[1]: Leaving directory '/lib/modules/3.16.0-4-amd64/build'
Makefile:1049: recipe for target 'modules' failed
make: *** [modules] Error 2

joseph-jja, 2016-02-06:
You need the kernel headers and kernel built toolsd. Check the Debian website for instruction on how to install the right stuff.

boinkitsbroken, 2017-01-26:
@joseph-jja,

If he needed the kernel headers he wouldn't have a directory called "linux-headers-3.16.0.4-amd64" now would he?

The truth is that the origininal makefile for this driver is garbage. We need to write a new one that works properly with GNU Make. I'm working on it.

joseph-jja, 2017-01-26:
@boinkitsbroken  I said kernel headers and kernel build tools, if you read it says no rule to make modules, which means he is not setup to build a kernel.  which is why I pointed him debian website and he needs to install the correct stuff.  this is a kernel module and you must be setup to build a kernel module on your platform.  each linux platform is different.  

as far as the original makefile being garbage, we didn't write that makefile it was done by the hardware maker

boinkitsbroken, 2017-01-28:
@joseph-jja If you read it says "no rule to make modules" which is an error I have seen repeatedly despite having all the packages setup to build a kernel. In fact, I've built several kernels already using  my current setup for GNU make--but this driver still fails with the same error every time. This is why I said the makefile is garbage. Why else would the issues list contain numerous complaints of make errors?

The makefile is trash, it's been trash from the start, and it needs to be built again from scratch. For starters, it's clear that the manufacturer tried to create a "one-size-fits-all" driver for linux. I plan to get it working only on Debian for x86/x64 PCs, and even that will be challenge enough for a month or so with my current skills. I hope to have something workable to share soon.

Again, "no rule to make modules" does NOT mean he is not setup to build a kernel. The makefile is demonstrably faulty.

joseph-jja, 2017-01-28:
Again, "as far as the original makefile being garbage, we didn't write that makefile it was done by the **hardware maker**". If you can make it better then do so. I didn't write it.

boinkitsbroken, 2017-01-28:
I never said you did. I know the manufacturer is to blame here, and I have laid the blame squarely on them. I CAN make a better makefile, and I SHALL do so, given the time to do it. 

Now that we are in violent agreement, may I please continue my work?

rudra01, 2017-02-01:
hello, i am trying to use wiriless D-link DWA-131 (Bus 001 Device 003: ID 2001:3319 D-Link Corp.) but when i try to install the driver i get this error msg : 
root@debian:/tmp/DWA-131/20140812_rtl8192EU_linux_v4.3.1.1_11320# make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/3.16.0-4-amd64/build M=/tmp/DWA-131/20140812_rtl8192EU_linux_v4.3.1.1_11320  modules
make[1]: Entering directory '/lib/modules/3.16.0-4-amd64/build'
make[1]: *** No rule to make target 'modules'.  Stop.
make[1]: Leaving directory '/lib/modules/3.16.0-4-amd64/build'
Makefile:1323: recipe for target 'modules' failed
make: *** [modules] Error 2
 i have tried to fix it but i cant 
someone help me please ....