abperiasamy/rtl8812AU_8821AU_linux #113: modprobe insert fails

frank-dspeed, 2016-01-13:
[  498.711274] 8812au: disagrees about version of symbol module_layout
[  525.335027] 8812au: disagrees about version of symbol module_layout

btong, 2016-02-23:
I saw this with the upgrade from 3.13.0-77-generic to 3.13.0-79-generic

harshavardhana, 2016-02-23:
> I saw this with the upgrade from 3.13.0-77-generic to 3.13.0-79-generic (ubuntu trusty)

Every kernel upgrade you need to rebuild the module, or add dkms so that dkms does that for you.

btong, 2016-02-24:
I am using dkms, I have no wifi after the upgrade. Manually running modprobe gives the error above.

harshavardhana, 2016-02-24:
> I am using dkms, I have no wifi after the upgrade. Manually running modprobe gives the error above.

Some times even with that you have to rebuild, dkms could have failed.

closed 2016-09-11 by harshavardhana