abperiasamy/rtl8812AU_8821AU_linux #116: Compile and AP mode on Raspberry Pi 2

Polioman, 2016-01-26:
Hi,

I use this driver with an Edimax EW-7811UAC AC600 dongle. I'm under Raspbian Jessie with kernel 4.1.13.
I follow instructions from Readme, I can install all prerequisites without issue but when it comes to this step, It doesn't work :

Append following at the end of your /boot/config.txt, reboot your Pi

`kernel=vmlinuz-3.10-3-rpi`
`initramfs initrd.img-3.10-3-rpi followkernel`

I've tried with 3.6 and 3.18.0 values since I've got these files in /boot instead of 3.10-3, my RPi doesn't boot.

Make and make install work without appending these lines but I've got an issue when I try to put my dongle in 802.11ac AP mode :

`Could not set channel for kernel driver`
`Interface initialization failed`

It works in 802.11a mode.

I don't understand what I'm doing wrong.

joseph-jja, 2016-02-06:
On the raspberry pi I found you have to build the kernel from source first and then build the module. It takes some time to built a kernel on the pi though.

Polioman, 2016-02-08:
I found a working driver here : https://www.raspberrypi.org/forums/viewtopic.php?f=36&t=134207