abperiasamy/rtl8812AU_8821AU_linux #121: having trouble compiling on ubuntu mate running  4.2.0-35-generic

cjwsam, 2016-03-30:
having trouble compiling on ubuntu mate running 

4.2.0-35-generic

# sudo make -j1 V=1
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.2.0-35-generic/build M=/var/lib/dkms/rtl8812AU_8821AU_linux/1.0/build  modules
make[1]: Entering directory '/usr/src/linux-headers-4.2.0-35-generic'
test -e include/generated/autoconf.h -a -e include/config/auto.conf || (		\
echo >&2;							\
echo >&2 "  ERROR: Kernel configuration is invalid.";		\
echo >&2 "         include/generated/autoconf.h or include/config/auto.conf are missing.";\
echo >&2 "         Run 'make oldconfig && make prepare' on kernel src to fix it.";	\
echo >&2 ;							\
/bin/false)
mkdir -p /var/lib/dkms/rtl8812AU_8821AU_linux/1.0/build/.tmp_versions ; rm -f /var/lib/dkms/rtl8812AU_8821AU_linux/1.0/build/.tmp_versions/*
make -f ./scripts/Makefile.build obj=/var/lib/dkms/rtl8812AU_8821AU_linux/1.0/build
(cat /dev/null; ) > /var/lib/dkms/rtl8812AU_8821AU_linux/1.0/build/modules.order
make -f ./scripts/Makefile.modpost
  find /var/lib/dkms/rtl8812AU_8821AU_linux/1.0/build/.tmp_versions -name '*.mod' | xargs -r grep -h '\.ko$' | sort -u | sed 's/\.ko$/.o/' | scripts/mod/modpost -m -a -i ./Module.symvers -I /var/lib/dkms/rtl8812AU_8821AU_linux/1.0/build/Module.symvers  -o /var/lib/dkms/rtl8812AU_8821AU_linux/1.0/build/Module.symvers -S -w  -s -T -
make[1]: Leaving directory '/usr/src/linux-headers-4.2.0-35-generic'

harshavardhana, 2016-03-30:
> make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.2.0-35-generic/build M=/var/lib/dkms/rtl8812AU_8821AU_linux/1.0/build modules
> make[1]: Entering directory '/usr/src/linux-headers-4.2.0-35-generic' [...]

This seems some kernel source related issue.

cjwsam, 2016-03-30:
Yea but is there a work around or anything i can do to fix it ?

harshavardhana, 2016-03-30:
> Yea but is there a work around or anything i can do to fix it ?

Not sure at all, perhaps re-installing the kernel sources and attempting to compile the driver again.