abperiasamy/rtl8812AU_8821AU_linux #122: Cannot compile/install accordingly to the published details.

notorand-it, 2016-03-30:
I have followed the details given in the README.md (which is also displayed the main page of this project) "Compiling with DKMS".
The `sudo dkms build -m rtl8812AU_8821AU_linux -v 1.0` command fails with:

```
Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
'make'....
ERROR (dkms apport): binary package for rtl8812AU_8821AU_linux: 1.0 not found
Error!  Build of 8812au.ko failed for: 4.5.0-040500-lowlatency (x86_64)
Consult the make.log in the build directory
/var/lib/dkms/rtl8812AU_8821AU_linux/1.0/build/ for more information.
```

while the cited `make.log` contains:

```
DKMS make.log for rtl8812AU_8821AU_linux-1.0 for kernel 4.5.0-040500-lowlatency (x86_64)
mer 30 mar 2016, 11.26.41, CEST
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.5.0-040500-lowlatency/build M=/var/lib/dkms/rtl8812AU_8821AU_linux/1.0/build  modules
make[1]: Entering directory `/usr/src/linux-headers-4.5.0-040500-lowlatency'
  Building modules, stage 2.
  MODPOST 0 modules
make[1]: Leaving directory `/usr/src/linux-headers-4.5.0-040500-lowlatency'
```

These last lines are also printed out when running `make` in the main directory where the `git clone` command pulled the files.

harshavardhana, 2016-03-30:
> Kernel preparation unnecessary for this kernel.  Skipping...
>
> Building module:
> cleaning build area....
> 'make'....
> ERROR (dkms apport): binary package for rtl8812AU_8821AU_linux: 1.0 not found [...]

Have you tried doing simple 'make' and 'make install'?

notorand-it, 2016-03-30:
@harshavardhana Yes, of course. It's written in my report, lower lines.

blthong, 2016-03-30:
I am also facing problems installing this driver.  I am very new to Linux, any help would be much appreciated.  In my case, it appears the file 8812au.ko could not be found.  The following are the results when I tried 'make' and 'make install':

thong-450-031d rtl8812AU_8821AU_linux # make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/3.19.0-32-generic/build M=/rtl8812AU_8821AU_linux  modules
make[1]: Entering directory `/usr/src/linux-headers-3.19.0-32-generic'
  Building modules, stage 2.
  MODPOST 0 modules
make[1]: Leaving directory `/usr/src/linux-headers-3.19.0-32-generic'
thong-450-031d rtl8812AU_8821AU_linux # sudo make install
install -p -m 644 8812au.ko  /lib/modules/3.19.0-32-generic/kernel/drivers/net/wireless/
install: cannot stat ‘8812au.ko’: No such file or directory
make: *** [install] Error 1

harshavardhana, 2016-03-30:
> @harshavardhana Yes, of course. It's written in my report, lower lines.

Strange will see.. if its a current master bug.

kerail, 2016-04-01:
I have the same problem with raspberry pi.
Seems that this commit break the thing:
https://github.com/abperiasamy/rtl8812AU_8821AU_linux/commit/5f1739bdffd9dab3361e8c9813df8f68b4437a3b

random-thoughts, 2016-04-01:
Same problem as bithong for kernel 4.2

notorand-it, 2016-04-01:
@kerail Have you tried to revert that single change? What are the outcomes?

mkreisl, 2016-04-01:
@notorand-it 
I did it, but does not work

harshavardhana, 2016-04-01:
> @notorand-it 
> I did it, but does not work
> 
> Hmmm, strange. Built kernel from scratch, revert commit 5f1739b and worked again
> 
> So I can confirm: this commit is the culprit

Reverting it. Thanks

harshavardhana, 2016-04-01:
Can everyone validate the latest master? thank you.

mkreisl, 2016-04-01:
Nothing new there. Did you forgot to push?

harshavardhana, 2016-04-01:
> Nothing new there. Did you forgot to push?

Ah you are right :P - done now.

closed 2016-04-04 by harshavardhana