abperiasamy/rtl8812AU_8821AU_linux #124: RTL8814?

DanMan, 2016-03-31:
I see mention of that chipset in the sources, but if I plug in a [D-Link DWA-192](https://wikidevi.com/wiki/D-Link_DWA-192), the driver currently doesn't seem to support it, unfortunatly.

Any idea what's missing? The firmware maybe?

djpaek, 2016-04-01:
unsubscribe

harshavardhana, 2016-04-01:
> I see mention of that chipset in the sources, but if I plug in a D-Link DWA-192, the driver currently doesn't seem to support it, unfortunatly.
> 
> Any idea what's missing? The firmware maybe?

Not sure, never really tested it was a community patch.

DanMan, 2016-04-03:
Well, what would I need to do to test it? I'm about to send it back, so...

harshavardhana, 2016-04-04:
> Well, what would I need to do to test it? I'm about to send it back, so...

when you say it doesn't support it means that kernel is not able to detect it? are there any messages that you see in 'dmesg' ? 

The problem could be firmware, so validating it would be hard.

DanMan, 2016-04-04:
Yes. dmesg says the same as without the driver installed. I've tried compiling the other 8812au drivers floating around here on github, to no avail.

nicholas5, 2017-09-12:
+1 for same problem with TP Link T9U :(

ptpt52, 2017-09-12:
show your dmesg when plug in the device

DanMan, 2018-07-04:
https://github.com/diederikdehaas/rtl8814AU

closed 2018-07-04 by DanMan