abperiasamy/rtl8812AU_8821AU_linux #126: EV3DEV

Tom-Shen, 2016-04-08:
Hello, Sir

If I want to use it in EV3DEV, how should I change the makefile?

Thanks.

harshavardhana, 2016-04-08:
> If I want to use it in EV3DEV, how should I change the makefile?

Do not understand what you mean? you need to change the code to support it if its a new device.

Tom-Shen, 2016-04-08:
I believe in the makefile I need to change the below section:
==============================================
CONFIG_PLATFORM_I386_PC = y 
53 CONFIG_PLATFORM_ANDROID_X86 = n 
54 CONFIG_PLATFORM_JB_X86 = n 
55 CONFIG_PLATFORM_ARM_S3C2K4 = n 
56 CONFIG_PLATFORM_ARM_PXA2XX = n 
57 CONFIG_PLATFORM_ARM_S3C6K4 = n 
58 CONFIG_PLATFORM_ARM_RPI = n 
59 CONFIG_PLATFORM_MIPS_RMI = n 
60 CONFIG_PLATFORM_RTD2880B = n 
61 CONFIG_PLATFORM_MIPS_AR9132 = n 
62 CONFIG_PLATFORM_RTK_DMP = n 
63 CONFIG_PLATFORM_MIPS_PLM = n 
64 CONFIG_PLATFORM_MSTAR389 = n 
65 CONFIG_PLATFORM_MT53XX = n 
66 CONFIG_PLATFORM_ARM_MX51_241H = n 
67 CONFIG_PLATFORM_FS_MX61 = n 
68 CONFIG_PLATFORM_ACTIONS_ATJ227X = n 
69 CONFIG_PLATFORM_TEGRA3_CARDHU = n 
70 CONFIG_PLATFORM_TEGRA4_DALMORE = n 
71 CONFIG_PLATFORM_ARM_TCC8900 = n 
72 CONFIG_PLATFORM_ARM_TCC8920 = n 
73 CONFIG_PLATFORM_ARM_TCC8920_JB42 = n 
74 CONFIG_PLATFORM_ARM_RK2818 = n 
75 CONFIG_PLATFORM_ARM_URBETTER = n 
76 CONFIG_PLATFORM_ARM_TI_PANDA = n 
77 CONFIG_PLATFORM_MIPS_JZ4760 = n 
78 CONFIG_PLATFORM_DMP_PHILIPS = n 
79 CONFIG_PLATFORM_TI_DM365 = n 
80 CONFIG_PLATFORM_MSTAR_TITANIA12 = n 
81 CONFIG_PLATFORM_SZEBOOK = n 
82 CONFIG_PLATFORM_ARM_SUNxI = n 
83 CONFIG_PLATFORM_ARM_SUN6I = n 
84 CONFIG_PLATFORM_ACTIONS_ATM702X = n 
85 CONFIG_PLATFORM_ACTIONS_ATV5201 = n 

==============================================

which one I should use?

Tom-Shen, 2016-04-08:
Lego EV3 is using ARM9 so which option I should use?