abperiasamy/rtl8812AU_8821AU_linux #127: The frame size of 1048 bytes is larger than 1024 bytes

notorand-it, 2016-04-14:
I am getting this warning.
```
  CC [M]  /home/enzo/rtl8812AU_8821AU_linux/os_dep/linux/ioctl_cfg80211.o
/home/enzo/rtl8812AU_8821AU_linux/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_inform_bss’:
/home/enzo/rtl8812AU_8821AU_linux/os_dep/linux/ioctl_cfg80211.c:474:1: warning: the frame size of 1048 bytes is larger than 1024 bytes [-Wframe-larger-than=]
 }
 ^

```
Please, consider a code fix or a compilation one.

harshavardhana, 2016-04-15:
Good idea looking for code contributions.

joseph-jja, 2016-11-10:
Pull the latest code.  I saw a similar message and pulled the latest and it is gone.

coderofsalvation, 2022-06-01:
I have some more info about this in error.<br>
In case you're still getting this message:<br><br><Br>

```
float buf[4096]={0};   // look for these kind of declarations
```
then rewrite to:
```
#define SIZE 64 // smaller amount      // alternatively use malloc()
float wav_data[SIZE]={0};                   //
```

I think cosmopolitan puts some restraints on size to ensure it works on as much platforms as possible.