abperiasamy/rtl8812AU_8821AU_linux #129: Does this driver support Ubuntu 16.04 LTS?

daniel-dong, 2016-05-29:
Hi, I'm using this driver on Ubuntu 15.10 and it works very well. I'd like to upgrade my Ubuntu to 16.04 LTS, but not sure whether this driver supports it.

Thanks.

harshavardhana, 2016-05-29:
> Hi, I'm using this driver on Ubuntu 15.10 and it works very well. I'd like to upgrade my Ubuntu to 16.04 LTS, but not sure whether this driver supports it.
> 
> Thanks.

We haven't tested this yet on 16.04

grandcat, 2016-05-31:
Works for me on 16.04 with kernel `4.4.0-22-generic #40-Ubuntu SMP`.

HERBdotSYSTEMS, 2016-06-16:
I'm running 4.4.0-24-generic and it does *not* work.

daniel-dong, 2016-06-16:
Did you install the driver on a previous kernel using dkms before upgrading the kernel to 4.4.0-24? The problem might be caused by dkms or the driver's dkms configuration.

daniel-dong, 2016-06-27:
It works for my RTL8811AU on Ubuntu 16.04 (4.4.0-24-generic).

dgrat, 2016-06-27:
Works with a Debian RT kernel vs. 4.4

decosse, 2016-07-25:
Does not work for me - see https://github.com/abperiasamy/rtl8812AU_8821AU_linux/issues/138

I also found this - https://github.com/gnab/rtl8812au/issues/80 - that suggests Network Manager from that kernel seems to be the cause

alexkart, 2016-09-04:
I've installed Ubuntu 16.04
and run these commands
```
# sudo cp -R . /usr/src/rtl8812AU_8821AU_linux-1.0
# sudo dkms add -m rtl8812AU_8821AU_linux -v 1.0
# sudo dkms build -m rtl8812AU_8821AU_linux -v 1.0
# sudo dkms install -m rtl8812AU_8821AU_linux -v 1.0
```
then rebooted the system and nothing happened, do I need to run something else to get my Archer T2U working?

daniel-dong, 2016-09-04:
Did you upgrade the Linux kernel before rebooting the system?

alexkart, 2016-09-04:
Yes. I think I'm trying to use wrong driver, this driver is for Archer T4U (AC1200), right? But I need driver for Archer T2U (AC600), do you know where can I found it?

closed 2017-07-07 by ptpt52