abperiasamy/rtl8812AU_8821AU_linux #134: Compilation issue for kernel higher than 4.5

sukanyach, 2016-06-20:
Hi,

I am trying to compile this driver on ubuntu with kernel version higher than 4.5 (4.5.0-040500-generic). But it fails on my system. 

Here I am attaching my error message for reference.

Error -  "Cannot use CONFIG_CC_STACKPROTECTOR_STRONG: -fstack-protector-strong not supported by compiler"

harshavardhana, 2016-06-20:
We have never tested this on 4.5 kernel, feel free to send a patch @sukanyach

nharia, 2016-08-19:
Hi,
Am trying to compile this driver on kali linux kernel 4.0.0-kali1-amd64. this is the error i get

root@kali:~/Downloads/rtl8812AU_8821AU_linux-master# dkms build -m rtl8812AU_8821AU_linux -v 1.0
Error! echo
Your kernel headers for kernel 4.0.0-kali1-amd64 cannot be found at
/lib/modules/4.0.0-kali1-amd64/build or /lib/modules/4.0.0-kali1-amd64/source.

There is no build or source folder in the /lib/modules/4.0.0 folder.

harshavardhana, 2016-09-11:
> root@kali:~/Downloads/rtl8812AU_8821AU_linux-master# dkms build -m rtl8812AU_8821AU_linux -v 1.0
> Error! echo
> Your kernel headers for kernel 4.0.0-kali1-amd64 cannot be found at
> /lib/modules/4.0.0-kali1-amd64/build or /lib/modules/4.0.0-kali1-amd64/source.

you need to install kernel headers @nharia most probably 

```
apt install linux-headers
```

harshavardhana, 2016-09-11:
> Error - "Cannot use CONFIG_CC_STACKPROTECTOR_STRONG: -fstack-protector-strong not supported by compiler"

This looks like a compiler error more than the driver bug here..

dullroar, 2016-11-24:
I don't see a space in the path to the project above, but if there is one, it will cause this very error. Really. I was tearing my hair out until I saw the answer about spaces in the path to the project causing issues [here](http://stackoverflow.com/questions/12501124/errors-in-compiling-linux-kernel-module-config-x86-x32-enabled-but-no-binutils/27916315#27916315). Sounds stupid, I know. FWIW, this was on Linux Mint 18, 4.4.0.47 X86_64 kernel, gcc v5.4.0.