abperiasamy/rtl8812AU_8821AU_linux #137: (Sometimes) fails to reconnect after wake from suspend

Tinche, 2016-06-27:
Using Ubuntu 16.04, kernel 4.6.0, sometimes after waking from suspend my TP-Link AC1200 T4U doesn't want to reconnect. Generally I have to unplug and replug the adapter.

This keeps showing up in the logs:

```
[35229.038921] RTL871X: set ssid [LopiBalti] fw_state=0x00000009
[35229.125837] RTL871X: start auth
[35229.130303] RTL871X: auth success, start assoc
[35229.134767] RTL871X: assoc success
[35229.134842] ------------[ cut here ]------------
[35229.134923] WARNING: CPU: 4 PID: 27167 at /home/kernel/COD/linux/net/wireless/sme.c:714 __cfg80211_connect_result+0x395/0x400 [cfg80211]
[35229.134928] Modules linked in: joydev ipt_MASQUERADE nf_nat_masquerade_ipv4 iptable_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 xt_addrtype iptable_filter ip_tables xt_conntrack x_tables nf_nat nf_conntrack br_netfilter bridge stp llc 8812au(OE) cfg80211 snd_hda_codec_hdmi nvidia_uvm(POE) kvm_amd kvm snd_hda_codec_realtek snd_hda_codec_generic snd_hda_intel irqbypass snd_hda_codec snd_hda_core snd_hwdep snd_pcm serio_raw edac_mce_amd input_leds snd_seq_midi edac_core k10temp snd_seq_midi_event snd_rawmidi snd_seq snd_seq_device snd_timer snd soundcore i2c_piix4 shpchp 8250_fintek wmi mac_hid binfmt_misc overlay parport_pc ppdev lp parport autofs4 pata_acpi hid_generic usbhid hid nvidia_drm(POE) nvidia_modeset(POE) nvidia(POE) firewire_ohci drm_kms_helper psmouse syscopyarea firewire_core sysfillrect
[35229.135017]  sysimgblt crc_itu_t fb_sys_fops atl1c drm pata_atiixp ahci nvme libahci nvme_core fjes
[35229.135039] CPU: 4 PID: 27167 Comm: kworker/u16:76 Tainted: P        W  OE   4.6.0-040600-generic #201605151930
[35229.135045] Hardware name: To Be Filled By O.E.M. To Be Filled By O.E.M./870 Extreme3 R2.0, BIOS P1.80 12/06/2011
[35229.135087] Workqueue: cfg80211 cfg80211_event_work [cfg80211]
[35229.135092]  0000000000000286 00000000781dc25a ffff8804254bbca0 ffffffff813f1dd3
[35229.135101]  0000000000000000 0000000000000000 ffff8804254bbce0 ffffffff810827eb
[35229.135107]  000002cabab91c78 ffff8800bab91c00 0000000000000000 0000000000000000
[35229.135114] Call Trace:
[35229.135125]  [<ffffffff813f1dd3>] dump_stack+0x63/0x90
[35229.135134]  [<ffffffff810827eb>] __warn+0xcb/0xf0
[35229.135142]  [<ffffffff8108291d>] warn_slowpath_null+0x1d/0x20
[35229.135190]  [<ffffffffc0f015a5>] __cfg80211_connect_result+0x395/0x400 [cfg80211]
[35229.135232]  [<ffffffffc0ed871a>] cfg80211_process_wdev_events+0x13a/0x1b0 [cfg80211]
[35229.135273]  [<ffffffffc0ed871a>] ? cfg80211_process_wdev_events+0x13a/0x1b0 [cfg80211]
[35229.135315]  [<ffffffffc0ed87c2>] cfg80211_process_rdev_events+0x32/0x70 [cfg80211]
[35229.135354]  [<ffffffffc0ed301e>] cfg80211_event_work+0x1e/0x30 [cfg80211]
[35229.135363]  [<ffffffff8109c7a4>] process_one_work+0x164/0x470
[35229.135371]  [<ffffffff8109cafb>] worker_thread+0x4b/0x4f0
[35229.135378]  [<ffffffff8109cab0>] ? process_one_work+0x470/0x470
[35229.135385]  [<ffffffff810a31b8>] kthread+0xd8/0xf0
[35229.135393]  [<ffffffff818423e2>] ret_from_fork+0x22/0x40
[35229.135399]  [<ffffffff810a30e0>] ? kthread_create_on_node+0x1c0/0x1c0
[35229.135404] ---[ end trace 31eff9565799e067 ]---
[35229.136539] UpdateHalRAMask8812A => mac_id:0, networkType:0x03, mask:0x00000fff
               	 ==> rssi_level:0, rate_bitmap:0x00000ff5
[35230.086433] UpdateHalRAMask8812A => mac_id:0, networkType:0x03, mask:0x00000fff
               	 ==> rssi_level:2, rate_bitmap:0x00000ff0
```

Before you say patches welcome, I don't think I can figure this one out myself, but I can test any code you think might fix the problem. Hopefully the stack trace is any indication?

VelizarHristov, 2016-10-04:
Same problem on Arch Linux. This wasn't the case previously.

ptpt52, 2016-10-04:
@VelizarHristov 
hi, can you try this code out?
https://github.com/ptpt52/rtl8812au

VelizarHristov, 2016-10-07:
@ptpt Sorry, I've already swapped out my WiFi USB with another one and am no longer relying on this program.

paneq, 2016-10-12:
For me: 

```
[465354.080217] RTL871X: start auth
[465354.082567] RTL871X: auth success, start assoc
[465354.084919] RTL871X: assoc success
[465354.084992] ------------[ cut here ]------------
[465354.085056] WARNING: CPU: 2 PID: 12773 at /build/linux-aA_g14/linux-3.13.0/net/wireless/sme.c:662 __cfg80211_connect_result+0x3db/0x420 [cfg80211]()
[465354.085080] Modules linked in: 8812au(OX) rndis_host cdc_ether usbnet mii hid_generic usbhid hid btrfs raid6_pq xor ufs qnx4 hfsplus hfs minix ntfs msdos jfs xfs libcrc32c ath9k_htc ath9k_common ath9k_hw ath sunrpc veth xfrm_user xfrm_algo xt_addrtype xt_conntrack aufs ctr ccm xt_CHECKSUM iptable_mangle ipt_MASQUERADE iptable_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 nf_nat nf_conntrack xt_tcpudp bridge stp llc iptable_filter ip_tables x_tables rfcomm bnep binfmt_misc intel_rapl x86_pkg_temp_thermal intel_powerclamp coretemp snd_hda_codec_conexant kvm_intel uvcvideo joydev kvm videobuf2_vmalloc arc4 videobuf2_memops videobuf2_core snd_hda_intel rtl8192ce rtl_pci videodev rtlwifi snd_hda_codec serio_raw rtl8192c_common mac80211 snd_hwdep btusb snd_pcm bluetooth cfg80211 snd_page_alloc snd_seq_midi snd_seq_midi_event snd_rawmidi snd_seq snd_seq_device thinkpad_acpi snd_timer nvram snd mac_hid lpc_ich mei_me soundcore mei shpchp parport_pc ppdev lp parport nls_iso8859_1 dm_crypt crct10dif_pclmul crc32_pclmul aesni_intel aes_x86_64 lrw gf128mul glue_helper ablk_helper cryptd i915 psmouse ahci sdhci_pci libahci sdhci i2c_algo_bit drm_kms_helper e1000e drm ptp pps_core wmi video
[465354.085232] CPU: 2 PID: 12773 Comm: kworker/u16:65 Tainted: G        W  OX 3.13.0-95-generic #142-Ubuntu
[465354.085236] Hardware name: LENOVO 4180HU1/4180HU1, BIOS 83ET76WW (1.46 ) 07/05/2013
[465354.085267] Workqueue: cfg80211 cfg80211_event_work [cfg80211]
[465354.085272]  0000000000000000 ffff88010bdddca8 ffffffff81729ec6 0000000000000000
[465354.085279]  0000000000000009 ffff88010bdddce0 ffffffff8106987d 0000000000000000
[465354.085285]  0000000000000000 ffff88007dcd2618 ffff8801388dd000 ffff88001456e000
[465354.085292] Call Trace:
[465354.085305]  [<ffffffff81729ec6>] dump_stack+0x64/0x82
[465354.085312]  [<ffffffff8106987d>] warn_slowpath_common+0x7d/0xa0
[465354.085316]  [<ffffffff8106995a>] warn_slowpath_null+0x1a/0x20
[465354.085342]  [<ffffffffa031558b>] __cfg80211_connect_result+0x3db/0x420 [cfg80211]
[465354.085363]  [<ffffffffa02f2844>] cfg80211_process_wdev_events+0x194/0x1c0 [cfg80211]
[465354.085382]  [<ffffffffa02f28b0>] cfg80211_process_rdev_events+0x40/0x90 [cfg80211]
[465354.085400]  [<ffffffffa02ee01e>] cfg80211_event_work+0x1e/0x30 [cfg80211]
[465354.085406]  [<ffffffff81086158>] process_one_work+0x178/0x470
[465354.085411]  [<ffffffff81086f71>] worker_thread+0x121/0x410
[465354.085417]  [<ffffffff81086e50>] ? rescuer_thread+0x430/0x430
[465354.085421]  [<ffffffff8108dd59>] kthread+0xc9/0xe0
[465354.085426]  [<ffffffff8108dc90>] ? kthread_create_on_node+0x1c0/0x1c0
[465354.085432]  [<ffffffff8173a9a8>] ret_from_fork+0x58/0x90
[465354.085436]  [<ffffffff8108dc90>] ? kthread_create_on_node+0x1c0/0x1c0
[465354.085476] ---[ end trace 1091c516c31b3016 ]---
[465354.086932] UpdateHalRAMask8812A => mac_id:0, networkType:0x14, mask:0x0ffffff0
[465354.086932] 	 ==> rssi_level:0, rate_bitmap:0x0f8ff010
[465355.021011] UpdateHalRAMask8812A => mac_id:0, networkType:0x14, mask:0x0ffffff0
[465355.021011] 	 ==> rssi_level:2, rate_bitmap:0x0f8ff000
```

ptpt52, 2016-10-12:
@paneq 
seems your code is not the latest HEAD

paneq, 2016-10-13:
@ptpt52 ah, you are right indeed, That's because on latest I can't compile

```
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/3.13.0-95-generic/build M=/home/rupert/develop/paneq/abperiasamy_rtl8812AU_8821AU_linux.git  modules
make[1]: Entering directory `/usr/src/linux-headers-3.13.0-95-generic'
  CC [M]  /home/rupert/develop/paneq/abperiasamy_rtl8812AU_8821AU_linux.git/core/rtw_br_ext.o
In file included from include/net/ip.h:31:0,
                 from /home/rupert/develop/paneq/abperiasamy_rtl8812AU_8821AU_linux.git/core/rtw_br_ext.c:24:
include/net/route.h: In function ‘ip_route_output’:
include/net/route.h:124:3: error: missing initializer for field ‘flowic_iif’ of ‘struct flowi_common’ [-Werror=missing-field-initializers]
   .flowi4_tos = tos,
   ^
In file included from include/linux/netfilter.h:277:0,
                 from include/net/netns/netfilter.h:5,
                 from include/net/net_namespace.h:20,
                 from include/linux/netdevice.h:43,
                 from include/uapi/linux/if_arp.h:26,
                 from include/linux/if_arp.h:27,
                 from /home/rupert/develop/paneq/abperiasamy_rtl8812AU_8821AU_linux.git/core/rtw_br_ext.c:23:
include/net/flow.h:16:6: note: ‘flowic_iif’ declared here
  int flowic_iif;
      ^
In file included from include/net/ip.h:31:0,
                 from /home/rupert/develop/paneq/abperiasamy_rtl8812AU_8821AU_linux.git/core/rtw_br_ext.c:24:
include/net/route.h:125:3: error: missing initializer for field ‘flowic_scope’ of ‘struct flowi_common’ [-Werror=missing-field-initializers]
   .daddr = daddr,
   ^
In file included from include/linux/netfilter.h:277:0,
                 from include/net/netns/netfilter.h:5,
                 from include/net/net_namespace.h:20,
                 from include/linux/netdevice.h:43,
                 from include/uapi/linux/if_arp.h:26,
                 from include/linux/if_arp.h:27,
                 from /home/rupert/develop/paneq/abperiasamy_rtl8812AU_8821AU_linux.git/core/rtw_br_ext.c:23:
include/net/flow.h:19:7: note: ‘flowic_scope’ declared here
  __u8 flowic_scope;
       ^
cc1: all warnings being treated as errors
make[2]: *** [/home/rupert/develop/paneq/abperiasamy_rtl8812AU_8821AU_linux.git/core/rtw_br_ext.o] Error 1
make[1]: *** [_module_/home/rupert/develop/paneq/abperiasamy_rtl8812AU_8821AU_linux.git] Error 2
make[1]: Leaving directory `/usr/src/linux-headers-3.13.0-95-generic'
make: *** [modules] Error 2
```

paneq, 2016-10-13:
@ptpt52 FYI:

```
(master) > uname -a
Linux rupert-ThinkPad-T420 3.13.0-95-generic #142-Ubuntu SMP Fri Aug 12 17:00:09 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux

(master) > lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 14.04.5 LTS
Release:	14.04
Codename:	trusty
```

Should I create a new issue for that?

paneq, 2016-10-13:
I tried to use kernel 4.4 and had the same problem.

ptpt52, 2016-10-16:
@paneq 
try to build it without option: EXTRA_CFLAGS += -Werror

paneq, 2016-10-16:
@ptpt52 Thanks, that worked! I will see how it works tomorrow.

closed 2016-11-04 by ptpt52