abperiasamy/rtl8812AU_8821AU_linux #138: Ubuntu 14.04 ( 4.2.0-42) successfull installation, device not recognized

random-thoughts, 2016-07-05:
I am running Ubuntu 14.04 (Linux-image 4.2.0-42). The installation completes successfully with no error messages but the device is not visible to the system. I have tried plugging it out and in again as well as restarting the machine.

decosse, 2016-07-24:
My Asus AC56 had been working great with the previous kernel (Linux Mint 17.3) 

I just upgraded to Linux Mint 18 with new Kernel 4.4.0-21 and cannot get the driver operational 
I tried the latest clone and built with dkms
I get the following when I query lshw -class network 

htpc rtl8812AU_8821AU_linux # lshw -class network
  *-network               
       description: Ethernet interface
       product: Ethernet Connection I218-V
       vendor: Intel Corporation
       physical id: 19
       bus info: pci@0000:00:19.0
       logical name: eth0
       version: 04
       serial: ec:a8:6b:ff:7c:4d
       capacity: 1Gbit/s
       width: 32 bits
       clock: 33MHz
       capabilities: pm msi bus_master cap_list ethernet physical tp 10bt 10bt-fd 100bt 100bt-fd 1000bt-fd autonegotiation
       configuration: autonegotiation=on broadcast=yes driver=e1000e driverversion=3.2.6-k firmware=0.6-4 latency=0 link=no multicast=yes port=twisted pair
       resources: irq:44 memory:f7c00000-f7c1ffff memory:f7c3c000-f7c3cfff ioport:f080(size=32)
  *-network
       description: Wireless interface
       physical id: 2
       bus info: usb@2:3
       logical name: wlan0
       serial: ac:22:0b:94:8b:33
       capabilities: ethernet physical wireless
       configuration: broadcast=yes driver=rtl8812au multicast=yes wireless=unassociated

It's showing the device as wlan0 with correct MAC address and does show the rtl8812au driver
Suggestions?

decosse, 2016-07-25:
downgrading the Network Manager resolved this issue 
see - https://github.com/gnab/rtl8812au/issues/81

random-thoughts, 2016-11-29:
Hi!

The issue was the that the driver has an unsigned key and could not be loaded. A quick workaround is to use `sudo mokutil --disable-validation`, reboot the machine and enter the information the computer asks in the boot screen. Another solution is to disable secure boot. More info about it [here](http://askubuntu.com/questions/755238/why-disabling-secure-boot-is-enforced-policy-when-installing-3rd-party-modules)

About `mokutil`: When you issue the command, you are asked to insert a password. In the very beginning of the boot the computer will ask either part of that password (e.g. letters 4,6, and 5) or the whole password. Be quick because if the option disappears the driver will not load. In such case you should just reissue the command

closed 2016-11-29 by random-thoughts