abperiasamy/rtl8812AU_8821AU_linux #141: Device not recognized in Ubuntu Server Edition

ericychoi, 2016-08-02:
Under Ubuntu 14.04.2 LTS (GNU/Linux 3.16.0-77-generic x86_64) the device doesn't seem be recognized.
I followed the instruction here: http://swkstudios.com/tutorials/ubuntu/ubuntu-14-04-installing-tp-link-ac1200-t4u/

When I unplug and plug the device, I get the following from `dmesg`
```
[ 1008.518724] usb 1-1: new high-speed USB device number 3 using ehci-pci
[ 1008.651626] usb 1-1: New USB device found, idVendor=2357, idProduct=0101
[ 1008.651642] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1008.651650] usb 1-1: Product: 802.11n NIC
[ 1008.651658] usb 1-1: Manufacturer: Realtek
[ 1008.651665] usb 1-1: SerialNumber: 123456
```
.. and nothing else following.

Any suggestions?

ericychoi, 2016-08-02:
Realized I didn't put the exact model for the device: It's Archer T4U

http://www.tp-link.com/en/products/details/Archer-T4U.html

ptpt52, 2016-11-04:
fixed

closed 2016-11-04 by ptpt52