abperiasamy/rtl8812AU_8821AU_linux #144: Doesn't work well with  TP Link Archer T4U AC1200

TomBZombie, 2016-08-12:
I got two of these as I read that linux support worked, however I'm having an issue on each PC that uses it

PC 1: Intel i7 5557u, with the dongle attached the PC will freeze intermittently and require rebooting. The whole PC freezes, I can't switch to a TTY or SSH into it. Arch Linux, kernel 4.7. `journalctl -b -1` doesn't show anything unusual so it freezes before being able to log anything useful.

PC 2: Intel Xeon X5670, it doesn't freeze but it does seem to break systemd. After 10-15 mins the connection will drop and reconnect. Eventually networkmanager seems to crash and doesn't list all the networks.  A bigger issue is that it seems to break systemd entirely. After the connection goes down this appears in dmesg:

```
[ 7876.783743] RTL871X: linked_status_chk(wlp0s29f7u2) disconnect or roaming
[ 7878.808176] RTL871X: SetHwReg8812A:(HW_VAR_CHECK_TXBUF)NOT empty(177) in 2000 ms
[ 7878.808181] RTL871X: SetHwReg8812A:(HW_VAR_CHECK_TXBUF)0x200=0x00d61010, 0x204=0x00a40000 (0x00d61010, 0x00a40000)
[ 7885.663270] RTL871X: SetHwReg8812A:(HW_VAR_CHECK_TXBUF)NOT empty(177) in 2000 ms
[ 7885.663274] RTL871X: SetHwReg8812A:(HW_VAR_CHECK_TXBUF)0x200=0x00d61010, 0x204=0x00540000 (0x00d61010, 0x00540000)
[ 7890.518204] RTL871X: There are some pkts to transmit
[ 7890.518209] RTL871X: free_xmitbuf_cnt: 4, free_xmit_extbuf_cnt: 31
[ 7892.610221] RTL871X: SetHwReg8812A:(HW_VAR_CHECK_TXBUF)NOT empty(177) in 2000 ms
[ 7892.610225] RTL871X: SetHwReg8812A:(HW_VAR_CHECK_TXBUF)0x200=0x00d61010, 0x204=0x00050000 (0x00d61010, 0x00050000)
[ 7897.463826] RTL871X: There are some pkts to transmit
[ 7897.463830] RTL871X: free_xmitbuf_cnt: 4, free_xmit_extbuf_cnt: 0
[ 7899.474151] RTL871X: There are some pkts to transmit
[ 7899.474155] RTL871X: free_xmitbuf_cnt: 4, free_xmit_extbuf_cnt: 0
[ 7901.477565] RTL871X: There are some pkts to transmit
[ 7901.477569] RTL871X: free_xmitbuf_cnt: 4, free_xmit_extbuf_cnt: 0
[ 7904.473184] RTL871X: SetHwReg8812A:(HW_VAR_CHECK_TXBUF)NOT empty(178) in 2010 ms
[ 7904.473189] RTL871X: SetHwReg8812A:(HW_VAR_CHECK_TXBUF)0x200=0x00d61010, 0x204=0x00050000 (0x00d61010, 0x00050000)
[ 7905.816664] IPv6: ADDRCONF(NETDEV_UP): wlp0s29f7u2: link is not ready
[ 7909.333422] RTL871X: There are some pkts to transmit
[ 7909.333427] RTL871X: free_xmitbuf_cnt: 4, free_xmit_extbuf_cnt: 0
[ 7911.337644] RTL871X: There are some pkts to transmit
[ 7911.337649] RTL871X: free_xmitbuf_cnt: 4, free_xmit_extbuf_cnt: 0
[ 7913.343288] RTL871X: There are some pkts to transmit
[ 7913.343292] RTL871X: free_xmitbuf_cnt: 4, free_xmit_extbuf_cnt: 0
[ 7915.349967] RTL871X: There are some pkts to transmit
[ 7915.349972] RTL871X: free_xmitbuf_cnt: 4, free_xmit_extbuf_cnt: 0
[ 7917.356148] RTL871X: There are some pkts to transmit
[ 7917.356152] RTL871X: free_xmitbuf_cnt: 4, free_xmit_extbuf_cnt: 0
[ 7919.362978] RTL871X: There are some pkts to transmit
[ 7919.362982] RTL871X: free_xmitbuf_cnt: 4, free_xmit_extbuf_cnt: 0
[ 7921.372761] RTL871X: There are some pkts to transmit
[ 7921.372765] RTL871X: free_xmitbuf_cnt: 4, free_xmit_extbuf_cnt: 0
[ 7923.375819] RTL871X: There are some pkts to transmit
[ 7923.375823] RTL871X: free_xmitbuf_cnt: 4, free_xmit_extbuf_cnt: 0
[ 7925.382500] RTL871X: There are some pkts to transmit
[ 7925.382504] RTL871X: free_xmitbuf_cnt: 4, free_xmit_extbuf_cnt: 0
[ 7927.388935] RTL871X: There are some pkts to transmit
[ 7927.388942] RTL871X: free_xmitbuf_cnt: 4, free_xmit_extbuf_cnt: 0
[ 7929.395612] RTL871X: There are some pkts to transmit
[ 7929.395616] RTL871X: free_xmitbuf_cnt: 4, free_xmit_extbuf_cnt: 0
[ 7932.894317] RTL871X: SetHwReg8812A:(HW_VAR_CHECK_TXBUF)NOT empty(177) in 2006 ms
[ 7932.894322] RTL871X: SetHwReg8812A:(HW_VAR_CHECK_TXBUF)0x200=0x00d61010, 0x204=0x00050000 (0x00d61010, 0x00050000)
```

And trying to do anything with systemd no longer works:

```
sudo systemctl status NetworkManager
Failed to get properties: Connection timed out
sudo systemctl status httpd
Failed to get properties: Connection timed out
```

TomBZombie, 2016-08-12:
I've been streaming a tv stream to test this and it may be to do with running two. So far with just one in the second PC I haven't had a problem.

TomBZombie, 2016-08-12:
It's not interferrence. I had the Xeon PC with the dongle plugged in and the i7 without. After an hour and a half:

```
[   11.201769] RTL871X: nolinked power save enter
[   32.748158] RTL871X: nolinked power save leave
[   37.577511] RTL871X: nolinked power save enter
[   65.760658] RTL871X: nolinked power save leave
[   70.586508] RTL871X: nolinked power save enter
[  108.771887] RTL871X: nolinked power save leave
[  113.591862] RTL871X: nolinked power save enter
[  161.780946] RTL871X: nolinked power save leave
[  166.600170] RTL871X: nolinked power save enter
[  224.785708] RTL871X: nolinked power save leave
[  229.595059] RTL871X: nolinked power save enter
[  287.764723] RTL871X: nolinked power save leave
[  292.589698] RTL871X: nolinked power save enter
[  318.672955] IPv6: ADDRCONF(NETDEV_UP): wlp0s29f7u2: link is not ready
[  319.341292] RTL871X: nolinked power save leave
[  321.980322] Bluetooth: RFCOMM TTY layer initialized
[  321.980328] Bluetooth: RFCOMM socket layer initialized
[  321.980332] Bluetooth: RFCOMM ver 1.11
[  324.308905] RTL871X: nolinked power save enter
[  328.756388] RTL871X: nolinked power save leave
[  328.756631] RTL871X: rtw_set_802_11_connect(wlp0s29f7u2)  fw_state=0x00000008
[  328.889451] RTL871X: start auth
[  329.189933] RTL871X: auth success, start assoc
[  329.192077] RTL871X: rtw_cfg80211_indicate_connect(wlp0s29f7u2) BSS not found !!
[  329.192087] RTL871X: assoc success
[  329.192191] IPv6: ADDRCONF(NETDEV_CHANGE): wlp0s29f7u2: link becomes ready
[  329.195299] RTL871X: recv eapol packet
[  329.195816] RTL871X: send eapol packet
[  329.200543] RTL871X: recv eapol packet
[  329.200658] RTL871X: send eapol packet
[  329.206306] RTL871X: set pairwise key camid:4, addr:34:97:f6:51:7b:c4, kid:0, type:AES
[  329.208637] RTL871X: set group key camid:5, addr:34:97:f6:51:7b:c4, kid:1, type:AES
[  379.055900] RTL871X: rtw_set_802_11_connect(wlp0s29f7u2)  fw_state=0x00000008
[  379.268658] RTL871X: start auth
[  379.270772] RTL871X: auth success, start assoc
[  379.273038] RTL871X: rtw_cfg80211_indicate_connect(wlp0s29f7u2) BSS not found !!
[  379.273045] RTL871X: assoc success
[  379.275909] RTL871X: recv eapol packet
[  379.276439] RTL871X: send eapol packet
[  379.281156] RTL871X: recv eapol packet
[  379.281220] RTL871X: send eapol packet
[  379.287802] RTL871X: set pairwise key camid:4, addr:34:97:f6:51:7b:c4, kid:0, type:AES
[  379.290060] RTL871X: set group key camid:5, addr:34:97:f6:51:7b:c4, kid:1, type:AES
[  709.704193] RTL871X: recv eapol packet
[  709.704483] RTL871X: set group key camid:6, addr:34:97:f6:51:7b:c4, kid:2, type:AES
[  709.704483] RTL871X: send eapol packet
[ 1694.387031] RTL871X: linked_status_chk(wlp0s29f7u2) disconnect or roaming
[ 1696.979030] RTL871X: SetHwReg8812A:(HW_VAR_CHECK_TXBUF)NOT empty(177) in 2000 ms
[ 1696.979035] RTL871X: SetHwReg8812A:(HW_VAR_CHECK_TXBUF)0x200=0x00d61010, 0x204=0x00bd1000 (0x00d61010, 0x00bd1000)
[ 1701.816436] RTL871X: nolinked power save enter
[ 1795.378556] RTL871X: nolinked power save leave
[ 1800.202937] RTL871X: nolinked power save enter
[ 1915.337636] RTL871X: nolinked power save leave
[ 1920.158362] RTL871X: nolinked power save enter
```

This time it didn't kill systemd, I think that happend when I tried to reconnect after it disconnected the first time. 

I think the problem is with 5ghz only. I'm going to try leaving it connected to the 2.4ghz band and see if it stays connected.

The router is an Asus RT-AC3200 if it makes any difference.

joelabair, 2016-09-01:
I too am experiencing the "connection drops and reconnects" symptom as well with kernel 4.6.7 and the TP-Link T4U v1.0

The only remedy I've found is to unplug and replug the device. Which gives me another 10 min or so of connectivity.

hyph3n, 2016-10-01:
I am also trying to use the TP-Link T4U device without any luck. I believe mine is v1.0 (as I see CA/1.0 after the serial number). On one occasion, I experienced something similar to what @TomBZombie reported where the system froze completely and had to be physically reset while I was trying to connect.

The bigger issue, however, is that I have not been able to get it to connect to either of two different access points.

Before cloning and installing this repo, with the T4U plugged in, NetworkManager shows me no wireless adapters (and therefore no available networks). After cloning and installing this repo with DKMS, with the T4U plugged in I see a wireless adapter available, and the expected list of wireless networks.

Here are the different failed connections I've attempted:

1) 5ghz (AC) network with WPA2 security.

2) 5ghz (AC) network (on a different AP from step 1) with no wireless security.

3) 5ghz (N) network (on same AP as step 2)with no wireless security.

4) 5ghz (A) network (on same AP as step 2) with no wireless security.

5) 2.4ghz (N) network with WPA2 security.

6) 2.4ghz (N) network (on same AP as step 2) with no wireless security.

7) 2.4ghz (G) network (on same AP as step 2) with no wireless security.

8) 2.4ghz (B) network (on same AP as step 2) with no wireless security.

I've tried the unplug/replug that @joelabair suggested above, as well as reboots of the PC. I simply cannot get the T4U to connect to a wireless network.

A TP-Link TL-WN722N device works "out-of-the-box" (ie. no special driver required) in all 2.4ghz tests described above, so the problem doesn't appear to be specific to my OS configuration or my access points (which are both Linksys WRT1900AC, one running OpenWRT, the other on stock firmware).

Fedora 24
Linux 4.7.5-200.fc24.x86_64

Let me know if I can provide additional information that would be helpful.

ptpt52, 2016-10-02:
Maybe you can try out this code:
[https://github.com/ptpt52/rtl8812au
](url)

hyph3n, 2016-10-02:
Hi @ptpt52 

I cloned that repo and installed via "sudo make -f Makefile.dkms install", but unfortunately, the driver doesn't seem to recognize my device. I've tried three different USB ports, and get the same end result of not being able to see any wireless devices in NetworkManager. Here are some logs.

dmesg: 
> usb 2-2: new high-speed USB device number 5 using ehci-pci
> usb 2-2: New USB device found, idVendor=2357, idProduct=0101
> usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> usb 2-2: Product: 802.11n NIC
> usb 2-2: Manufacturer: Realtek
> usb 2-2: SerialNumber: 123456

lsusb:
> Bus 002 Device 005: ID 2357:0101 

journalctl:
> kernel: usb 2-2: new high-speed USB device number 5 using ehci-pci
> kernel: usb 2-2: New USB device found, idVendor=2357, idProduct=0101
> kernel: usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> kernel: usb 2-2: Product: 802.11n NIC
> kernel: usb 2-2: Manufacturer: Realtek
> kernel: usb 2-2: SerialNumber: 123456
> mtp-probe[1979]: checking bus 2, device 5: "/sys/devices/pci0000:00/0000:00:1d.7/usb2/2-2"
> mtp-probe[1979]: bus: 2, device: 5 was not an MTP device

NetworkManager:
> Shows no available wireless device.

I went ahead and ran "sudo make -f Makefile.dkms uninstall" since it didn't seem to be working.

Any thoughts?

ptpt52, 2016-10-02:
@hyph3n 
I just updated the source code, support for TP-Link - Archer T4U has been added.

hyph3n, 2016-10-02:
Hi @ptpt52 

It looks like I'm already using the latest version of your code.

> $ git pull
> Already up-to-date.

ptpt52, 2016-10-02:
@hyph3n sorry
it is a bug in my code, it only supports *8812au but your device is *8821au (the code is not complete)
but now,
I just updated the code to version 4.3.14
it is full support for you device now(maybe)

try it out as you wish.

ptpt52, 2016-10-02:
@hyph3n 
forget to tell you:
you may need to remove the old version
sudo dkms remove -m rtl8812au -v 4.3.8_12175.20140902 --all

hyph3n, 2016-10-02:
Thanks for your attention to this, @ptpt52 !

Unfortunately, the driver still isn't working as expected.

1) Confirmed previous version was removed. 
> Error! There are no instances of module: rtl8812au
> 4.3.8_12175.20140902 located in the DKMS tree.

2) Updated repo.
> git pull 

3) Ran installation on updated repo.
> sudo make -f Makefile.dkms install

4) Rebooted.

This time NetworkManager showed me **two** instances of the wireless device instead of zero. Unfortunately, neither instance of the device showed any wireless networks, so I was still not able to initiate a connection.

5) Uninstalled driver.
> sudo make -f Makefile.dkms uninstall

ptpt52, 2016-10-02:
@hyph3n 

hi, try this patch and reinstall

`
diff --git a/Makefile b/Makefile
index 7ce2180..a665b3b 100644
--- a/Makefile
+++ b/Makefile
@@ -864,9 +864,9 @@ endif
 
 ifeq ($(CONFIG_PLATFORM_I386_PC), y)
 EXTRA_CFLAGS += -DCONFIG_LITTLE_ENDIAN
-EXTRA_CFLAGS += -DCONFIG_CONCURRENT_MODE
-EXTRA_CFLAGS += -DCONFIG_IOCTL_CFG80211 -DRTW_USE_CFG80211_STA_EVENT
-EXTRA_CFLAGS += -DCONFIG_P2P_IPS
+#EXTRA_CFLAGS += -DCONFIG_CONCURRENT_MODE
+#EXTRA_CFLAGS += -DCONFIG_IOCTL_CFG80211 -DRTW_USE_CFG80211_STA_EVENT
+#EXTRA_CFLAGS += -DCONFIG_P2P_IPS
 SUBARCH := $(shell uname -m | sed -e s/i.86/i386/)
 ARCH ?= $(SUBARCH)
 CROSS_COMPILE ?=

`

hyph3n, 2016-10-02:
Copied everything between (but not including) the ` marks. Pasted into a file called patch.diff. Put patch.diff into the root of the repository.
> git apply patch.diff
> fatal: corrupt patch at line 7

Opened the file in vim, and line 7 is: 
> ifeq ($(CONFIG_PLATFORM_I386_PC), y)

ptpt52, 2016-10-08:
@hyph3n 
re-check the new version please
https://github.com/ptpt52/rtl8812au

hyph3n, 2016-10-10:
I just had a chance to test the latest code, but I'm still seeing the same problem. I'm able to install the module, and NetworkManager starts showing a "wireless" option, but there are no networks listed.

I performed a dnf upgrade earlier, so I'm now on kernel 4.7.5-200.fc24.x86_64. Are there additional logs I can check, or steps you'd like me to attempt?

ptpt52, 2016-10-16:
@hyph3n 
try the latest code
maybe it fixed your problem

Grief, 2016-10-17:
@TomBZombie @joelabair @hyph3n If you guys don't mind to upgrade the kernel up to 4.8.1and feel adventurous, feel free to try: https://github.com/Grief/archer-t4u It can also work on older kernels but I cannot guarantee

aasshhiisshh, 2016-10-22:
This is not working for me as well. @hyph3n Yeah man I got the same error..

hyph3n, 2016-10-22:
Good news, @ptpt52, @Grief, @aasshhiisshh, @TomBZombie. I just had a chance to pull the latest code this morning, and the TP Link Archer T4U is now working as expected! I'm successfully connected to a 5ghz access point.

Distro: Fedora 24
Kernel: 4.7.7-200.fc24.x86_64

Thanks for everyone's help. If anyone is still having trouble getting this to work on a Fedora system, let me know and hopefully I can assist.

ptpt52, 2016-10-22:
@hyph3n 
opps but my code is a fork of https://github.com/abperiasamy/rtl8812AU_8821AU_linux
they are the same

hyph3n, 2016-10-22:
@ptpt52, ok I just cloned yours because that's the link you've been providing. I didn't realize your changes were already part of this repo as well.

closed 2016-11-04 by ptpt52