abperiasamy/rtl8812AU_8821AU_linux #145: AWUS036AC does not correspond to the monitor mode?

RLambert142, 2016-08-26:
I need of help. 

# git clone https://github.com/abperiasamy/rtl8812AU_8821AU_linux                                                    
Cloning into 'rtl8812AU_8821AU_linux'...
remote: Counting objects: 843, done.                                                                                                        
remote: Total 843 (delta 0), reused 0 (delta 0), pack-reused 843                                                                            
Receiving objects: 100% (843/843), 2.06 MiB | 523.00 KiB/s, done.
Resolving deltas: 100% (432/432), done.
Checking connectivity... done.

# cd ./rtl8812AU_8821AU_linux/

# make                                                                                        
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/3.16.0-4-amd64/build M=/root/rtl8821au/rtl8812AU_8821AU_linux  modules
make[1]: Entering directory '/usr/src/linux-headers-3.16.0-4-amd64'
Makefile:10: *** mixed implicit and normal rules: deprecated syntax
make[1]: Entering directory `/usr/src/linux-headers-3.16.0-4-amd64'
  CC [M]  /root/rtl8821au/rtl8812AU_8821AU_linux/core/rtw_cmd.o
  CC [M]  /root/rtl8821au/rtl8812AU_8821AU_linux/core/rtw_security.o
  CC [M]  /root/rtl8821au/rtl8812AU_8821AU_linux/core/rtw_debug.o
  CC [M]  /root/rtl8821au/rtl8812AU_8821AU_linux/core/rtw_io.o
  CC [M]  /root/rtl8821au/rtl8812AU_8821AU_linux/core/rtw_ioctl_query.o
  CC [M]  /root/rtl8821au/rtl8812AU_8821AU_linux/core/rtw_ioctl_set.o
  CC [M]  /root/rtl8821au/rtl8812AU_8821AU_linux/core/rtw_ieee80211.o
  CC [M]  /root/rtl8821au/rtl8812AU_8821AU_linux/core/rtw_mlme.o
  CC [M]  /root/rtl8821au/rtl8812AU_8821AU_linux/core/rtw_mlme_ext.o
  CC [M]  /root/rtl8821au/rtl8812AU_8821AU_linux/core/rtw_wlan_util.o
  CC [M]  /root/rtl8821au/rtl8812AU_8821AU_linux/core/rtw_vht.o
  CC [M]  /root/rtl8821au/rtl8812AU_8821AU_linux/core/rtw_pwrctrl.o
  CC [M]  /root/rtl8821au/rtl8812AU_8821AU_linux/core/rtw_rf.o
  CC [M]  /root/rtl8821au/rtl8812AU_8821AU_linux/core/rtw_recv.o
  CC [M]  /root/rtl8821au/rtl8812AU_8821AU_linux/core/rtw_sta_mgt.o
  CC [M]  /root/rtl8821au/rtl8812AU_8821AU_linux/core/rtw_ap.o
  CC [M]  /root/rtl8821au/rtl8812AU_8821AU_linux/core/rtw_xmit.o
  CC [M]  /root/rtl8821au/rtl8812AU_8821AU_linux/core/rtw_p2p.o
  CC [M]  /root/rtl8821au/rtl8812AU_8821AU_linux/core/rtw_tdls.o
  CC [M]  /root/rtl8821au/rtl8812AU_8821AU_linux/core/rtw_br_ext.o
  CC [M]  /root/rtl8821au/rtl8812AU_8821AU_linux/core/rtw_iol.o
  CC [M]  /root/rtl8821au/rtl8812AU_8821AU_linux/core/rtw_sreset.o
  CC [M]  /root/rtl8821au/rtl8812AU_8821AU_linux/core/efuse/rtw_efuse.o
  CC [M]  /root/rtl8821au/rtl8812AU_8821AU_linux/os_dep/osdep_service.o
  CC [M]  /root/rtl8821au/rtl8812AU_8821AU_linux/os_dep/linux/os_intfs.o
  CC [M]  /root/rtl8821au/rtl8812AU_8821AU_linux/os_dep/linux/usb_intf.o
  CC [M]  /root/rtl8821au/rtl8812AU_8821AU_linux/os_dep/linux/usb_ops_linux.o
  CC [M]  /root/rtl8821au/rtl8812AU_8821AU_linux/os_dep/linux/ioctl_linux.o
  CC [M]  /root/rtl8821au/rtl8812AU_8821AU_linux/os_dep/linux/xmit_linux.o
  CC [M]  /root/rtl8821au/rtl8812AU_8821AU_linux/os_dep/linux/mlme_linux.o
  CC [M]  /root/rtl8821au/rtl8812AU_8821AU_linux/os_dep/linux/recv_linux.o
  CC [M]  /root/rtl8821au/rtl8812AU_8821AU_linux/os_dep/linux/ioctl_cfg80211.o
/root/rtl8821au/rtl8812AU_8821AU_linux/os_dep/linux/ioctl_cfg80211.c:5252:2: warning: initialization from incompatible pointer type [enabled by default]
  .change_station = cfg80211_rtw_change_station,
  ^
/root/rtl8821au/rtl8812AU_8821AU_linux/os_dep/linux/ioctl_cfg80211.c:5252:2: warning: (near initialization for ‘rtw_cfg80211_ops.change_station’) [enabled by default]
  CC [M]  /root/rtl8821au/rtl8812AU_8821AU_linux/os_dep/linux/rtw_android.o
  CC [M]  /root/rtl8821au/rtl8812AU_8821AU_linux/hal/hal_intf.o
  CC [M]  /root/rtl8821au/rtl8812AU_8821AU_linux/hal/hal_com.o
  CC [M]  /root/rtl8821au/rtl8812AU_8821AU_linux/hal/hal_com_phycfg.o
  CC [M]  /root/rtl8821au/rtl8812AU_8821AU_linux/hal/hal_phy.o
  CC [M]  /root/rtl8821au/rtl8812AU_8821AU_linux/hal/led/hal_usb_led.o
  CC [M]  /root/rtl8821au/rtl8812AU_8821AU_linux/hal/HalPwrSeqCmd.o
  CC [M]  /root/rtl8821au/rtl8812AU_8821AU_linux/hal/rtl8812a/Hal8812PwrSeq.o
  CC [M]  /root/rtl8821au/rtl8812AU_8821AU_linux/hal/rtl8812a/Hal8821APwrSeq.o
  CC [M]  /root/rtl8821au/rtl8812AU_8821AU_linux/hal/rtl8812a/rtl8812a_xmit.o
  CC [M]  /root/rtl8821au/rtl8812AU_8821AU_linux/hal/rtl8812a/rtl8812a_sreset.o
  CC [M]  /root/rtl8821au/rtl8812AU_8821AU_linux/hal/rtl8812a/rtl8812a_hal_init.o
  CC [M]  /root/rtl8821au/rtl8812AU_8821AU_linux/hal/rtl8812a/rtl8812a_phycfg.o
  CC [M]  /root/rtl8821au/rtl8812AU_8821AU_linux/hal/rtl8812a/rtl8812a_rf6052.o
  CC [M]  /root/rtl8821au/rtl8812AU_8821AU_linux/hal/rtl8812a/rtl8812a_dm.o
  CC [M]  /root/rtl8821au/rtl8812AU_8821AU_linux/hal/rtl8812a/rtl8812a_rxdesc.o
  CC [M]  /root/rtl8821au/rtl8812AU_8821AU_linux/hal/rtl8812a/rtl8812a_cmd.o
[... 81 lines cut ...]
# airmon-ng start wlan1                                                                                                        

PHY     Interface       Driver          Chipset

phy0    wlan1           ??????          Realtek Semiconductor Corp. 
cat: /sys/class/ieee80211/phy0/device/net/wlan1mon/type: No such file or directory.

Newly created monitor mode interface wlan1mon is *NOT* in monitor mode.
Removing non-monitor wlan1mon interface...

WARNING: unable to start monitor mode, please run "airmon-ng check kill"

# airmon-ng check kill

# airmon-ng start wlan1                                                                                                        

PHY     Interface       Driver          Chipset

phy0    wlan1           ??????          Realtek Semiconductor Corp. 
cat: /sys/class/ieee80211/phy0/device/net/wlan1mon/type: No such file or directory.

Newly created monitor mode interface wlan1mon is *NOT* in monitor mode.
Removing non-monitor wlan1mon interface...

WARNING: unable to start monitor mode, please run "airmon-ng check kill"

yuri-sevatz, 2017-01-04:
Pretty sure this is a problem with all versions of the driver (Including 4.3.22-beta).

The created interface doesn't show up in device/net/*, that's why the airmon-ng script fails.  I'm still learning about the underlying Kernel APIs here, but I'm not 100% sure the driver actually creates nodes in /sys/class/.../$PHY/device/net/* .

However, after executing the iw command that airmon-ng calls to run thsi check, the "iw dev" command seems to disagree with airmon-ng here -- in that the interface is actually defined, and shows up as the right type.

Not sure you'd want to chance running this driver with multiple phy's though 👎 .  Best just use the oldschool "iwconfig <iface> mode monitor".

ptpt52, 2017-07-07:
fixed

closed 2017-07-07 by ptpt52